vshymanskyy/TinyGSM #535: HTTPS Support for SIM7600

biswaKL, 2021-05-05:
Any support for SIM7600 HTTPS?

SRGDamia1, 2021-05-05:
Not at present.

ashwanisihag, 2021-05-28:
Here is support for HTTPS

See below code.

ashwanisihag, 2021-05-28:
/**
   @file       TinyGsmClientSIM7600.h
   @author     Volodymyr Shymanskyy
   @license    LGPL-3.0
   @copyright  Copyright (c) 2016 Volodymyr Shymanskyy
   @date       Nov 2016
*/

#ifndef SRC_TINYGSMCLIENTSIM7600_H_
#define SRC_TINYGSMCLIENTSIM7600_H_

// #define TINY_GSM_DEBUG Serial
// #define TINY_GSM_USE_HEX

#define TINY_GSM_MUX_COUNT 10
#define TINY_GSM_BUFFER_READ_AND_CHECK_SIZE

#include "TinyGsmBattery.tpp"
#include "TinyGsmGPRS.tpp"
#include "TinyGsmGPS.tpp"
#include "TinyGsmGSMLocation.tpp"
#include "TinyGsmModem.tpp"
#include "TinyGsmSMS.tpp"
#include "TinyGsmTCP.tpp"
#include "TinyGsmTemperature.tpp"
#include "TinyGsmTime.tpp"

#define GSM_NL "\r\n"
static const char GSM_OK[] TINY_GSM_PROGMEM    = "OK" GSM_NL;
static const char GSM_ERROR[] TINY_GSM_PROGMEM = "ERROR" GSM_NL;
#if defined       TINY_GSM_DEBUG
static const char GSM_CME_ERROR[] TINY_GSM_PROGMEM = GSM_NL "+CME ERROR:";
static const char GSM_CMS_ERROR[] TINY_GSM_PROGMEM = GSM_NL "+CMS ERROR:";
#endif

enum RegStatus {
  REG_NO_RESULT    = -1,
  REG_UNREGISTERED = 0,
  REG_SEARCHING    = 2,
  REG_DENIED       = 3,
  REG_OK_HOME      = 1,
  REG_OK_ROAMING   = 5,
  REG_UNKNOWN      = 4,
};

class TinyGsmSim7600 : public TinyGsmModem<TinyGsmSim7600>,
  public TinyGsmGPRS<TinyGsmSim7600>,
  public TinyGsmTCP<TinyGsmSim7600, TINY_GSM_MUX_COUNT>,
  public TinyGsmSMS<TinyGsmSim7600>,
  public TinyGsmGSMLocation<TinyGsmSim7600>,
  public TinyGsmGPS<TinyGsmSim7600>,
  public TinyGsmTime<TinyGsmSim7600>,
  public TinyGsmBattery<TinyGsmSim7600>,
  public TinyGsmTemperature<TinyGsmSim7600> {
    friend class TinyGsmModem<TinyGsmSim7600>;
    friend class TinyGsmGPRS<TinyGsmSim7600>;
    friend class TinyGsmTCP<TinyGsmSim7600, TINY_GSM_MUX_COUNT>;
    friend class TinyGsmSMS<TinyGsmSim7600>;
    friend class TinyGsmGPS<TinyGsmSim7600>;
    friend class TinyGsmGSMLocation<TinyGsmSim7600>;
    friend class TinyGsmTime<TinyGsmSim7600>;
    friend class TinyGsmBattery<TinyGsmSim7600>;
    friend class TinyGsmTemperature<TinyGsmSim7600>;

    /*
       Inner Client
    */
  public:
    class GsmClientSim7600 : public GsmClient {
        friend class TinyGsmSim7600;
[... 964 lines cut ...]
      String data;
      return waitResponse(timeout_ms, data, r1, r2, r3, r4, r5);
    }

    int8_t waitResponse(GsmConstStr r1 = GFP(GSM_OK),
                        GsmConstStr r2 = GFP(GSM_ERROR),
#if defined TINY_GSM_DEBUG
                        GsmConstStr r3 = GFP(GSM_CME_ERROR),
                        GsmConstStr r4 = GFP(GSM_CMS_ERROR),
#else
                        GsmConstStr r3 = NULL, GsmConstStr r4 = NULL,
#endif
                        GsmConstStr r5 = NULL) {
      return waitResponse(1000, r1, r2, r3, r4, r5);
    }

  public:
    Stream& stream;

  protected:
    GsmClientSim7600* sockets[TINY_GSM_MUX_COUNT];
    const char*       gsmNL = GSM_NL;
};

#endif  // SRC_TINYGSMCLIENTSIM7600_H_

ashwanisihag, 2021-05-28:
I got success in sending data to both aws and azure TLS 1.2...its going good.

erlo, 2021-06-12:
Hello @ashwanisihag,

Would you accept to provide your TinyGsmClientSIM7600.h in raw format ?
I'm trying to find something to support HTTPS on SIM7600, and I have difficulties to work with your file.

Do you implement the class TinyGsmClientSecure ?

Thanks

EmanueleFeola, 2021-06-14:
hi @ashwanisihag I'm trying to use your code but I can't make it work, could you please provide your raw file?

EmanueleFeola, 2021-06-17:
@ashwanisihag @erlo I made a fork of tinygsm and modified two files: TinyGsmClientSIM7600.h and TinyGsmClient.h, please check it out.
I looked through code above and I got it working, so thank you a lot @ashwanisihag !!
I'm using the sim7600 module to connect to a mosquitto broker using the secure port 8883.
I tested it and it's working correctly, the esp32 connects to mosquitto broker successfully.
In my case I do not use certificates on the esp32 so I commented the lines in the code.
In the arduino script I declare the client as TinyGsmClientSecure client(modem);

ashwanisihag, 2021-06-22:
I am glad that it worked.:-)

kiddtang, 2021-06-27:
I found a relative easy way to make it support HTTPS. Check out this [tutorial](https://dogcomp.medium.com/send-https-request-from-sim-7600x-lte-module-4f76be19e900)!

tradexsrl, 2021-09-10:
Hi @EmanueleFeola , i have the same problem you had with the same mqtt server configuration, i download from your repository the two files modified by you but when i compile i get exit status 1 error 
if i revert to the original files and set TinyGsmClient client(modem) i'm able to compile....i don't know what could be.
please let me know

tradexsrl, 2021-09-10:
i activated the verbose level and it says:
\Documents\Arduino\libraries\TinyGSM\src/TinyGsmClient.h:46:25: error: 'GsmClientSecureSIM7600' in 'class TinyGsmSim7600' does not name a type
 typedef TinyGsmSim7600::GsmClientSecureSIM7600 TinyGsmClientSecure;

tradexsrl, 2021-09-10:
ok found where is the problem: there is a SerialMon.print(c); left in the TinyGsmClientSIM7600.h downloaded from @EmanueleFeola tinygsm fork

tradexsrl, 2021-09-11:
At least I was able to compile the program , in the comeing days I'll test the program on lilygo board with MQTT on port 8883
If I have problems I will post a comment here
Thanks

georeb, 2023-06-20:
Hi @EmanueleFeola Thank you for this patch - it works nicely on an ESP32 :) but I have a question:
When making an HTTPS GET request, the request reliably works, but there is a strange delay.

Please see the output below:

```
Performing HTTPS GET request... [1063451] modemSend called 
[1063462] modemSend waitResponse> called 
[1063477] modemSend waitResponse CCHSEND called 
[1065489] modemSend called 
[1065490] modemSend waitResponse> called 
[1065498] modemSend waitResponse CCHSEND called 
[1067507] modemSend called 
[1067508] modemSend waitResponse> called 
[1067522] modemSend waitResponse CCHSEND called 
[1069534] modemSend called 
[1069535] modemSend waitResponse> called 
[1069549] modemSend waitResponse CCHSEND called 
[1071562] modemSend called 
[1071563] modemSend waitResponse> called 
[1071574] modemSend waitResponse CCHSEND called 
[1073583] modemSend called 
[1073584] modemSend waitResponse> called 
[1073594] modemSend waitResponse CCHSEND called 
[1075603] modemSend called 
[1075604] modemSend waitResponse> called 
[1075619] modemSend waitResponse CCHSEND called 
[1077633] modemSend called 
[1077634] modemSend waitResponse> called 
[1077645] modemSend waitResponse CCHSEND called 
[1079656] modemSend called 
[1079657] modemSend waitResponse> called 
[1079670] modemSend waitResponse CCHSEND called 
[1081682] modemSend called 
[1081683] modemSend waitResponse> called 
[1081697] modemSend waitResponse CCHSEND called 
[1083707] modemSend called 
[1083708] modemSend waitResponse> called 
[1083722] modemSend waitResponse CCHSEND called 
[1085735] modemSend called 
[1085736] modemSend waitResponse> called 
[1085749] modemSend waitResponse CCHSEND called 
[1087761] modemSend called 
[1087762] modemSend waitResponse> called 
[1087773] modemSend waitResponse CCHSEND called 
[1089783] modemSend called 
[1089784] modemSend waitResponse> called 
[1089796] modemSend waitResponse CCHSEND called 
[1091808] modemSend called 
[1091809] modemSend waitResponse> called 
[1091822] modemSend waitResponse CCHSEND called 
[1092129] modemGetAvailable   called 
[1092130] modemGetAvailable  +CCHRECV? called 
[1092142] modemGetAvailable  +CCHRECV?2 called 
[1092142] ### Available: 340 on 0
[1092142] modemRead  called 
[1092144] modemRead  CCHRECV called 
HTTP/1.1 200 OK
Server: nginx
Date: Tue, 20 Jun 2023 22:42:08 GMT
Content-Type: application/json
Transfer-Encoding: chunked
Connection: keep-alive
X-Powered-By: PHP/7.4.33
Access-Control-Allow-Origin: *
Access-Control-Allow-Headers: Content-Type
X-Powered-By: PleskLin

30
{"status":200,"status_message":"OK","data":null}
[... 91 lines cut ...]
[1093173] Dats is "api.mywebsite.com"
Response:
{"status":200,"status_message":"OK","data":null}
Body length is: 48
```

The library appears to be filling up the buffer (`CCHSEND`) with only a handful of bytes, every 2 secs and it takes approx 30 secs to finish filling the buffer.

Here is the recorded TX RX UART data, showing 30 seconds of `CCHSEND`:

[image]

Then it reads the buffer back(`CCHRECV`) all in one go, fine.
But then sends the following command 30 times, all receiving the same ERROR response from the SIM7600:

>`AT+CCHRECV=0,649\r\n`
`AT+CCHRECV=0,649\r\r\nERROR\r\n`

...executed 30 times in quick succession and then finishes with...

>`AT+CCHRECV?\r\n`
`AT+CCHRECV?\r\r\n+CCHRECV: LEN,0,0\r\n\r\nOK\r\n`

Why? It means that a simple HTTPS GET request is unnecessarily long.
Any insight would be much appreciated.