vshymanskyy/TinyGSM #538: Client sent an HTTP request to an HTTPS server Issue

mlrochaa23, 2021-05-06:
I'm working on a project to send data to Elasticsearch.

I was already able to send the data via wifi. When I try to do the same via GSM, I'm receiving " Client sent an HTTP request to an HTTPS server" message.

Below is part of the code:

`
TinyGsmClient client(modem);

 client.print(String("POST ") + url + " HTTP/1.1\r\n" +
                 "Host: " + host + "\r\n" +
                 "User-Agent: esp8266temp\r\n" +
                 "WWW-Authenticate: Basic realm=\"security\"\r\n" +
                 "Authorization: Basic " + userpass + "\r\n" +
                 "Connection: close\r\n" +
                 "Content-Type: application/json\r\n" +
                 "Content-Length: " + doc.length() + "\r\n\r\n" +
                 doc + "\r\n");
  
   // Serial.println("request sent");
    while (client.connected()) {
      String line = client.readStringUntil('\n');
      if (line == "\r") {
        SerialMon.println("headers received");
        break;
      }
    }
    String line = client.readStringUntil('\n');
    SerialMon.println("reply was:");
    SerialMon.println(line);
    delay(refresh_seconds * 10);
  } else {
    delay(10);
  }
   `
Any tip to solve this issue?

mlrochaa23, 2021-05-06:
``I'm working on a project to send data to Elasticsearch.

I was already able to send the data via wifi. When I try to do the same via GSM, I'm receiving " Client sent an HTTP request to an HTTPS server" message.

Below is part of the code:

`
TinyGsmClient client(modem);

 client.print(String("POST ") + url + " HTTP/1.1\r\n" +
                 "Host: " + host + "\r\n" +
                 "User-Agent: esp8266temp\r\n" +
                 "WWW-Authenticate: Basic realm=\"security\"\r\n" +
                 "Authorization: Basic " + userpass + "\r\n" +
                 "Connection: close\r\n" +
                 "Content-Type: application/json\r\n" +
                 "Content-Length: " + doc.length() + "\r\n\r\n" +
                 doc + "\r\n");
  
   // Serial.println("request sent");
    while (client.connected()) {
      String line = client.readStringUntil('\n');
      if (line == "\r") {
        SerialMon.println("headers received");
        break;
      }
    }
    String line = client.readStringUntil('\n');
    SerialMon.println("reply was:");
    SerialMon.println(line);
    delay(refresh_seconds * 10);
  } else {
    delay(10);
  }
   `
Any tip to solve this issue?``

SRGDamia1, 2021-05-11:
You need to use `TinyGsmClientSecure secureClient(modem);` not `TinyGsmClient client(modem);`  SSL isn't supported for all of the modules; check the ReadMe.

mlrochaa23, 2021-05-11:
The module is on firmware 14.18. 
I tried to run the simple example HTTPSClient to test the SSL but I had not success too. The module is able to connect to SIM provider, but when the module try to run the GET request, I'm receving "failed to connect" message.

SRGDamia1, 2021-05-11:
Your module probably just doesn't support SSL.  Search through the issues; others have had success using a standard (not-secured) TCP connection on the SIM800 with a secondary SSL library supplying the security layer.

woodlist, 2021-05-25:
> I tried that, but I had not success. When I use TinyGSMCLientSecure, I can't connect to elasticsearch server.
> I'm using the SIM800l module.

Try to use [https://github.com/OPEnSLab-OSU/SSLClient](url) You can eliminate the usage of TinyGSMCLientSecure

ashwanisihag, 2021-05-28:
It supports SSL

Here is full code

/**
   @file       TinyGsmClientSIM7600.h
   @author     Volodymyr Shymanskyy
   @license    LGPL-3.0
   @copyright  Copyright (c) 2016 Volodymyr Shymanskyy
   @date       Nov 2016
*/

#ifndef SRC_TINYGSMCLIENTSIM7600_H_
#define SRC_TINYGSMCLIENTSIM7600_H_

// #define TINY_GSM_DEBUG Serial
// #define TINY_GSM_USE_HEX

#define TINY_GSM_MUX_COUNT 10
#define TINY_GSM_BUFFER_READ_AND_CHECK_SIZE

#include "TinyGsmBattery.tpp"
#include "TinyGsmGPRS.tpp"
#include "TinyGsmGPS.tpp"
#include "TinyGsmGSMLocation.tpp"
#include "TinyGsmModem.tpp"
#include "TinyGsmSMS.tpp"
#include "TinyGsmTCP.tpp"
#include "TinyGsmTemperature.tpp"
#include "TinyGsmTime.tpp"

#define GSM_NL "\r\n"
static const char GSM_OK[] TINY_GSM_PROGMEM    = "OK" GSM_NL;
static const char GSM_ERROR[] TINY_GSM_PROGMEM = "ERROR" GSM_NL;
#if defined       TINY_GSM_DEBUG
static const char GSM_CME_ERROR[] TINY_GSM_PROGMEM = GSM_NL "+CME ERROR:";
static const char GSM_CMS_ERROR[] TINY_GSM_PROGMEM = GSM_NL "+CMS ERROR:";
#endif

enum RegStatus {
  REG_NO_RESULT    = -1,
  REG_UNREGISTERED = 0,
  REG_SEARCHING    = 2,
  REG_DENIED       = 3,
  REG_OK_HOME      = 1,
  REG_OK_ROAMING   = 5,
  REG_UNKNOWN      = 4,
};

class TinyGsmSim7600 : public TinyGsmModem<TinyGsmSim7600>,
  public TinyGsmGPRS<TinyGsmSim7600>,
  public TinyGsmTCP<TinyGsmSim7600, TINY_GSM_MUX_COUNT>,
  public TinyGsmSMS<TinyGsmSim7600>,
  public TinyGsmGSMLocation<TinyGsmSim7600>,
  public TinyGsmGPS<TinyGsmSim7600>,
  public TinyGsmTime<TinyGsmSim7600>,
  public TinyGsmBattery<TinyGsmSim7600>,
  public TinyGsmTemperature<TinyGsmSim7600> {
    friend class TinyGsmModem<TinyGsmSim7600>;
    friend class TinyGsmGPRS<TinyGsmSim7600>;
    friend class TinyGsmTCP<TinyGsmSim7600, TINY_GSM_MUX_COUNT>;
    friend class TinyGsmSMS<TinyGsmSim7600>;
    friend class TinyGsmGPS<TinyGsmSim7600>;
    friend class TinyGsmGSMLocation<TinyGsmSim7600>;
    friend class TinyGsmTime<TinyGsmSim7600>;
    friend class TinyGsmBattery<TinyGsmSim7600>;
    friend class TinyGsmTemperature<TinyGsmSim7600>;

    /*
       Inner Client
[... 968 lines cut ...]
      String data;
      return waitResponse(timeout_ms, data, r1, r2, r3, r4, r5);
    }

    int8_t waitResponse(GsmConstStr r1 = GFP(GSM_OK),
                        GsmConstStr r2 = GFP(GSM_ERROR),
#if defined TINY_GSM_DEBUG
                        GsmConstStr r3 = GFP(GSM_CME_ERROR),
                        GsmConstStr r4 = GFP(GSM_CMS_ERROR),
#else
                        GsmConstStr r3 = NULL, GsmConstStr r4 = NULL,
#endif
                        GsmConstStr r5 = NULL) {
      return waitResponse(1000, r1, r2, r3, r4, r5);
    }

  public:
    Stream& stream;

  protected:
    GsmClientSim7600* sockets[TINY_GSM_MUX_COUNT];
    const char*       gsmNL = GSM_NL;
};

#endif  // SRC_TINYGSMCLIENTSIM7600_H_

ashwanisihag, 2021-05-28:
I have tried with both AWS as well as azute iot...i got success in connecting both as TLS1.2
See the code above.
Regards

mlrochaa23, 2021-06-22:
> > I tried that, but I had not success. When I use TinyGSMCLientSecure, I can't connect to elasticsearch server.
> > I'm using the SIM800l module.
> 
> Try to use [https://github.com/OPEnSLab-OSU/SSLClient](url) You can eliminate the usage of TinyGSMCLientSecure

Hi woodlist. Many thanks for your tip. I tried that and worked fine!!

SRGDamia1, 2021-07-06:
If this is working for you now, could you please close the issue?

closed 2021-07-06 by mlrochaa23

ramonpaolo, 2024-02-17:
@mlrochaa23 share please