vshymanskyy/TinyGSM #539: random Reconnecting

warshaa, 2021-05-11:
Hi ,

I'm using the  esp32 with sim868 to connect mqtt broker. 

it connects to the broker and works normally however it keep disconnecting randomly every few seconds it connects back .

main loop :
while (1){
`    if (!mqtt.connected())
    {
        Serial.println("=== MQTT NOT CONNECTED ===");
        uint32_t t = millis();
        if (t - lastReconnectAttempt > 10000L)
        {
            lastReconnectAttempt = t;
            if (mqttConnect())
            {
                lastReconnectAttempt = 0;
            }
        }
        delay(1000);
        return;
    }
    mqtt.loop();`
    delay (100)
}

here is serial output with the debug enabled 

`AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,0,"TCP","3.248.172.59","1883","CONNECTED"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,0,"TCP","3.248.172.59","1883","CONNECTED"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,0,"TCP","3.248.172.59","1883","CONNECTED"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,0,"TCP","3.248.172.59","1883","CONNECTED"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,0,"TCP","3.248.172.59","1883","CONNECTED"

OK
AT+CIPRXGET=4,0

+CIPRPGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,0,"TCP","3.248.172.59","1883","CONNECTED"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,0,"TCP","3.240.172.59","1883","CONNECTED"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,0,"TCP","3.248.172.59","1883","CONNECTED"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,0,"TCP","3.248.172.59","1883","CONNECTED"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,0,"TCP","3.248.172.59","1883","CONNECTED"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,0,"TCP","3.248.172.59","1883","CONNECTED"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,0,"TCP","3.248.172.59","1883","CONNECTED"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,0,"TCP","3.248.172.59","1883","CONNECTED"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+AIPSTATUS: 0,0,"TCP","3.248.172.59","1883","CONNECTED"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,0,"TCP","3.248.172.59","1883","CONNECTED"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,0,"TCP","3.248.172.59","1883","CONNECTED"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,0,"TCP","3.248.172.59","1883","REMOTE CLOSING"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,0,"TCP","3.248.172.59","1883","REMOTE CLOSING"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTA␔US: 0,0,"TC�","3.248.172.59","1883","REMOTE CLOSING"

OK
AT+CIPCLOSE=0,1

0, CLOSE OK
 fail
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,0,"TCP","3.248.172.59","1883","CLOSED"

OK
=== MQTT NOT CONNECTED ===
`

warshaa, 2021-05-11:
I set the keep alive 600 seconds , still same thing

SRGDamia1, 2021-05-12:
Make the keep-alive shorter, not longer.  You want to ping the server more often to keep the connection open, not less often.

warshaa, 2021-05-15:
Hi Sara ,thanks for your reply.

I have tried different keepalive values but it seems that the connection will close in less than one minute if the keep alive is higher than 1 minute. 
as you mentioned decreasing the keep alive would solve the problem.
However with less than 1 minute keep alive the problem is the device would consume more power and data which are very limited in my situation, moreover the broker charges me on each packet transferred between the client and the broker including ping for keep alive.

warshaa, 2021-05-15:
problem solved !

the keep alive time has nothing wrong , the problem was in my setup function code which cause the tinyGsm client close the tcp connection every 1000ms.

thanks ,,

closed 2021-05-15 by warshaa