vshymanskyy/TinyGSM #540: LilyGO-T-SIM7600X fails to connect sometimes (+NETOPEN: 1)

gorghino, 2021-05-13:
[ X ] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[ X ] Bug or issue with library functionality (ie, sending data over TCP/IP)
[ ] Question or request for help

### What are you working with?

Modem: SIMCOM INCORPORATED SIMCOM_SIM7600E-H Revision: SIM7600M22_V1.1
Main processor board: ESP32-WROVER-B 
TinyGSM version: 0.10.9
Code: 
main.ino from Lilygo Producer: https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7600X/blob/master/examples/Arduino_TinyGSM/AllFunctions/AllFunctions.ino
with a couple of changes in the lib:

- I edited setNetworkMode() in order to work with Lilygo's loop (adjusting AllFunctions.ino is probably better):
```
String setNetworkMode(uint8_t mode) {
    sendAT(GF("+CNMP="), mode);
    if(waitResponse() == 1)
      return "OK";
    else
      return "";
  }
```
- I edited restartImpl() adding waitResponse on PB DONE as you did for SIM5360.h; it seems to connect easier with this fix.

```
  bool restartImpl(const char* pin = NULL) {
    if (!testAT()) { return false; }
    sendAT(GF("+CRESET"));
    if (waitResponse(10000L) != 1) { return false; }

    //delay(5000L);  // TODO(?):  Test this delay!

    // After booting, modem sends out messages as each of its
    // internal modules loads.  The final message is "PB DONE".
    if (waitResponse(40000L, GF(GSM_NL "PB DONE")) != 1) {
      return false;
    }
    return init(pin);
  }
```

### Scenario, steps to reproduce
I'm trying to connect my module to ThingsMobile network. The signal quality is very good (99) but sometimes I get +NETOPEN: 1 always followed by SMS DONE and a long timeout before restarting the loop.
In the test I reported on pastebin (links below) the second loop got connection. 
I suspect there're some wrong delays for my module or similar. 
I don't get why the first loop has two AT+CRESET with one OK/SMS DONE/PB DONE while the second has OK/SMS DONE(x2)/RDY/PB DONE and if it may conflict with the following connection phase.

I'm powering the module via USB (pin VIN as described here https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7600X/blob/master/README.MD) using a fast-charge USB-C cable.

### Expected result
Consistent connection phase

### Actual result
Sometimes NETOPEN fails

### Debug and AT command log
First loop - connection **fails**: https://pastebin.com/ruzzymeN
Second loop - connection OK: https://pastebin.com/VECGe2EY

gorghino, 2021-05-13:
I don't if it's related but it fails when`AT+CGREG?` gets `+CGREG: 0,5` from the beginning; the connection works when `AT+CGREG?` gets `0,0`, then `0,2 `and eventually `0,5`.

SRGDamia1, 2021-05-14:
I don't have a SIM7x00 to test myself.  My experience with all of the modules is that the connection process is really hard to get just right.  If you find something more stable, please submit a pull request.