vshymanskyy/TinyGSM #541: Quantity attempts

yaroshd81, 2021-05-17:
Where is the number of attempts to connect to the given host(ip) is indicated?

yaroshd81, 2021-05-17:
I know, what 20 tries connecting, but i can't find this variable or constant

SRGDamia1, 2021-05-17:
Usually it's just one.

yaroshd81, 2021-05-17:
Ok, but where this variable/constant? What is name variable/constant??? I can't find it.

SRGDamia1, 2021-05-17:
There is no variable or constant defining how many retries to make on a connection.  The `client.connect(...)` function passes off to the `modemConnect(...)` function which uses appropriate AT commands to open a TCP connection and then returns true/false based on whether the modem reported success or failure.  There isn't any retrying involved unless you've written that into your own code.  If the modem itself attempts a connection multiple times before returning a response, that's part of the modem's built in firmware, nothing this library does.

SRGDamia1, 2021-05-17:
One retry.  No variable or constant.  Just 1.

yaroshd81, 2021-05-17:
Ok, but after command:
```
mqtt_.connect
```
I have this log
```
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,,"","","","INITIAL"

OK
AT+CIPCLOSE=0,1

ERROR
AT+CIPSSL=0

OK
AT+CIPSTART=0,"TCP","mqtt.server",1883

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,0,"TCP","","1883","CONNECTING"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,0,"TCP","","1883","CONNECTING"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,0,"TCP","","1883","CONNECTING"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK

0, CONNECT FAIL
AT+CIPSTATUS=0

+CIPSTATUS: 0,,"","","","INITIAL"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,,"","","","INITIAL"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,,"","","","INITIAL"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,,"","","","INITIAL"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,,"","","","INITIAL"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,,"","","","INITIAL"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,,"","","","INITIAL"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,,"","","","INITIAL"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,,"","","","INITIAL"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,,"","","","INITIAL"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,,"","","","INITIAL"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,,"","","","INITIAL"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,,"","","","INITIAL"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,,"","","","INITIAL"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,,"","","","INITIAL"

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,,"","","","INITIAL"

OK
```
An attempt to connect takes 40 seconds, which is a very long time. And makes 20 identical requests (except for the initial one)

yaroshd81, 2021-05-17:
40 sec, because not pay. If paid - connecting the very fast, but we can have situation, when don't pay internet on our sim-card. (sorry for my bad english - i learn germany). Example for this situation, when no money: no internet, but sim-card active and GPRS connected.

yaroshd81, 2021-05-18:
I'm find it, sorry

closed 2021-05-18 by yaroshd81