vshymanskyy/TinyGSM #547: [HELP WANTED - ENHANCEMENT REQUEST] - Adding support for the NL-SWN-LTE-NRF9160

haimiko, 2021-05-26:
Looking to add support for the [NL-SWN-LET-NRF9160](https://nimbelink.com/Documentation/Skywire/Nano/4G_LTE_Cat_M1_Nordic/1002399_NL-SWN-LTE-NRF9160_AT-Command-Manual.pdf) .  I've had a stab at it.  I can get IMEI, SIMCCID,  other meta data but am having difficulty with the SOCKET connections.  Is anyone willing to take on this project.  Will pay someone to get it added and will share the results with the community.

SRGDamia1, 2021-05-26:
Have you seen https://github.com/vshymanskyy/TinyGSM/issues/486?

haimiko, 2021-05-27:
> 
> 
> Have you seen #486?

Yes but FW is different as well as AT commands.   I'd prefer to use the sock firmware and not have to flash the modem.

SRGDamia1, 2021-05-27:
This library is intended to be used on an external chip that communicates with a modem over a serial line; it is *NOT* intended to be used as part of the application firmware on a divided chip like the nRF9160.  Even if you were to try to use it to parse AT commands going between the user-programmable application firmware and secure modem firmware of the 9160, at the AT level the secure modem firmware does not support socket-level AT commands.  Look carefully through https://infocenter.nordicsemi.com/index.jsp?topic=%2Fug_nrf91_dk_gsg%2FUG%2Fnrf91_DK_gsg%2Fintro.html; there's nothing TCP or socket related at all.  The only way to open sockets from within the user programmable portion of the application firmware is to use the functions provided within the nRF SDK.

So, if you want to be able to use TinyGSM with the 9160, the *only* way would be to flash the user portion of your nRF9160 with something like Nordic's "serial LTE modem" application firmware and then to write a module for TinyGSM to be run on a second processor.  That's what https://github.com/aldras was working on in #486.

I have the nRF9160 development kit that I bought a while ago to test out, but I very quickly gave up on it.  For my use cases, trying to both modify and flash the nRF9160 itself and then write the TinyGSM wrapper for it was not worth the effort.

SRGDamia1, 2021-05-27:
This is similar to (but not quite the same as) a common confusion in using TinyGSM with an ESP8266 or an ESP32.  TinyGSM can be used within the firmware on either chip to communicate with an external cellular modem.  But TinyGSM cannot be used for wifi within a program running on an espressif chip.  The TinyGSM module for the ESP8266 only works if that espressif chip is the external modem and has been flashed with espressif's AT firmware.

haimiko, 2021-05-27:
Hi San, The modem firmware supplied by Nibelink does support non-secured
SOCKET and MQTT via AT commands (see
https://nimbelink.com/Documentation/Skywire/Nano/4G_LTE_Cat_M1_Nordic/1002399_NL-SWN-LTE-NRF9160_AT-Command-Manual.pdf)
.  I've already successfully done so with a terminal program.   Not my
intention to use TinyGSM on the internal modem chip.  Nordic provides a SDK
for that but not in scope for this request. I was  planning on using it on
an external MCU to make non-secure AT SOCKET and MQTT commands via UART1.
.    I was looking for help to get TinyGSM working with the stock SkyWire
firmware.

SRGDamia1, 2021-05-27:
I haven't looked at in depth, but I think the Skywire firmware is nothing more than the Nordic serial firmware.

haimiko, 2021-05-28:
Possibly however as I said I successfully initiated soctet and mqtt
communication via terminal programs.

Using this document
https://nimbelink.com/Documentation/Skywire/Nano/4G_LTE_Cat_M1_Nordic/1002400_NL-SW-LTE-NRF9160_SocketDialAppNote.pdf&ved=2ahUKEwiqr8vnjevwAhUDqJ4KHRaKBXkQFjABegQIJBAC&usg=AOvVaw1b6XA-2C4L9oh7M8YWgdbb

SRGDamia1, 2021-06-01:
As I said, I think the firmware Nimbelink uses is just the serial AT example that Nordic provides, which is nearly identical to that in #486

haimiko, 2021-06-01:
Nearly is correct but not identical enough to work.

aldras, 2022-01-17:
I have updated the forked repo with added support and (almost) stock firmware in #486.

https://github.com/aldras/TinyGSM

Please see the post in that thread for details.

The "almost" part is because I added support for HEX data (required for BearSSL or binary transfers).
By default it runs in HEX mode but by editing the src/TinyGsmClientNRF9160.h and changing the define options at the top of the file, this can be changed, and it will then work off pure stock v1.8.0 (latest as of this post) serial_lte_modem firmware.
#define DATATYPE 0 //HEX databode
//#define DATATYPE 1 //text databode

The SDK's serial_lte_modem sample application has changed, v1.6.1 was the last release (that I know of) that included the ability to send/receive either text data or HEX data depending on the command options issued.  This was removed and to be able to use that feature I added the commands #SENDB and #RECVB, as can be seen in the "src/TinyGsmClientNRF9160.h" file.

aldras, 2022-01-17:
I would like to comment a little about the AT command set and it has changed slightly as the SDK from Nordic does evolve, so the version 1.6.1 SDK will not exactly match the version 1.8.0 SDK.

I specifically had to copy some C code from the 1.6.1 serial_lte_modem to the 1.8.0 serial_lte_modem in order to add back in support for HEX send/receive commands.  This is why it is "nearly" stock, but flashing of the nRF9160 will be required as it does not arrive from the factory with the serial_lte_modem firmware installed.

However, if the define change is done above, it will run off the 100% unchanged serial_lte_modem sample application provided from Nordic in the v1.8.0 SDK.