vshymanskyy/TinyGSM #549: [other] SSL Client compatible with TinyGSM and ESP32 boards

Kromtar, 2021-05-30:
Hello everybody :)

I want to tell you that I have been testing [this library](https://github.com/govorox/SSLClient) which has turned out to be very useful to establish SSL connection using ESP32 based boards. I know there are many questions and problems with establishing secure connections using TinyGSM.
Clearly the solution will depend on your particular board, the model of the modem, the type of connection and several other factors.

In my case this library [easily solved](https://github.com/govorox/SSLClient) my problems. I have successfully tested it on ESP32 with SIM800L and SIM7000G.

I hope that for someone it may be useful.
Thank you maintainers and collaborators of TinyGSM.

nickagian, 2022-06-26:
Hi @Kromtar,

would you mind posting an example working code that combines the two libraries?
Because in my code I'm not able to run it successfully. I get a core panic error and it is rather difficult to debug.

Kromtar, 2022-06-26:
In the [examples folder](https://github.com/govorox/SSLClient/tree/master/examples/Esp32) you can find fully functional codes.

vshymanskyy, 2022-09-12:
Thank you. I think this may be worth mentioning in the README

closed 2022-09-12 by vshymanskyy

rtrombetta, 2022-12-02:
Hi @Kromtar!

I'm working on a project using the LiliGO T-SIM7000G board. Using the native functions of TinyGSM to conectar using SSL (TINY_GSM_MODEM_SIM7000SSL), despite not having any error during compilation, in my case the connections do not work. I tryed some exemples with https and mqtt. With the library you posted, everything worked perfectly defining the modem as TINY_GSM_MODEM_SIM7000. Both HTTPS and MQTT SSL connections on port 8883. Thank you very much for posting this library!

icabrera29, 2023-04-17:
Thank you very much

ilgarbenli, 2023-12-06:
I couldn't POST json request to googleapis.com would anybody done this before?

nh-ee, 2023-12-16:
I was trying to use this lib to connect to AWS S3 with API gateway, but no success till now. Did anyone work on such previously?

ilgarbenli, 2023-12-16:
@vshymanskyy Some SSL connections worked for me with this library: github.com/govorox/SSLClient

NavyaDevB, 2024-07-16:
> I'm working on a project using the LiliGO T-SIM7000G board. Using the native functions of TinyGSM to conectar using SSL (TINY_GSM_MODEM_SIM7000SSL), despite not having any error during compilation, [...]

>>>>>>>>>> 

> Hi @Kromtar!
> Can you tell me the code flow and libraries used to connect TTGO SIM7000G ESP32 Wrover-E to AWS i.e., with certificates using MQTT.
 
I have been trying for so many days, but I am getting many errors and I am unable to connected to AWS IoT core. It is reporting as MQTT connection failed, rc: -2