vshymanskyy/TinyGSM #551: SIM7000 not working since version 10.9

star297, 2021-06-02:
## What type of issues is this?

[x ] Bug or problem compiling the library
[x ] Bug or issue with library functionality (ie, sending data over TCP/IP)

### What are you working with?

Modem: SIM7000E, SIM7600E
Main processor board: ESP32
TinyGSM version: 11. 1,2,3,4
Code: 
```
 String result;
  do {
    result = modem.setNetworkMode(2);
    Serial.printf(".");
    delay(500);
  } while (result != "OK");
```

### Scenario, steps to reproduce
Can't get past the above function.
Working fine on version 10.9
I have the same problem with SIM7600E, was working on 10.9 and not since.

Is there a list of library changes that would effect the SIM7000 and SIM7600 since 10.9 ?
I can change code and check with the SIM7000, waiting for more SIM7600 modems atm.

gorghino, 2021-06-03:
setNetworkMode() doesn't return "OK" anymore but a boolean.
Some examples use the old setNetworkMode() but I guess you can edit the while condition to `while (!result)`

star297, 2021-12-06:
Version 0.11.5 is working on SIM7000 after the setNetworkMode() change.

closed 2021-12-06 by star297