vshymanskyy/TinyGSM #552: HttpsClient example's host doesn't work, but others do

itisluiz, 2021-06-08:
[x] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[x] Bug or issue with library functionality (ie, sending data over TCP/IP)
[ ] Question or request for help

### What are you working with?

Modem: SimCom Sim800 Revision: 1418B04SIM800L24
Main processor board: ESP32
TinyGSM version: 0.11.4
Code: examples\HttpsClient\HttpsClient.ino

### Scenario, steps to reproduce
Attempt to establish a secure connection to the host in the example ([vsh.pp.ua](url))

### Expected result
It works with Google, as expected.
```
AT+CIPSTART=0,"TCP","www.google.com",443

OK

OK

0, CONNECT OK
AT+CIPSEND=0,3

>GET 
DATA ACCEPT:0,3
AT+CIPSEND=0,1

>  
DATA ACCEPT:0,1
AT+CIPSEND=0,11

>/robots.txt 
DATA ACCEPT:0,11
AT+CIPSEND=0,9

> HTTP/1.1 
DATA ACCEPT:0,9
AT+CIPSEND=0,2
```

### Actual result
It doesn't work with the provided host within the exaple
```
AT+CIPSTART=0,"TCP","vsh.pp.ua",443

OK

OK

0, CLOSE OK
failed to connect
```

### What I tried
At first I assumed it had to be an issue regarding TLS, since the example using regular http worked fine for that same domain. So I tried explicitly disabling the certificate validation with "AT+SSLOPT=0,1", which didn't yield results. I then thought it could be related to the TLS version, the example host is running TLS v1.2 and google runs TLS v1.3. Trying a different website with TLS v1.2 works fine, just like google did.

### Debug and AT command log
```
Waiting for network...AT+CREG?

+CREG: 0,1

OK
 success
AT+CREG?

+CREG: 0,1

OK
Network connected
Connecting to timbrasil.brAT+CIPSHUT

SHUT OK
AT+CGATT=0

+SAPBR 1: DEACT

OK
AT+SAPBR=3,1,"Contype","GPRS"

OK
AT+SAPBR=3,1,"APN","timbrasil.br"

OK
AT+SAPBR=3,1,"USER","tim"

OK
AT+SAPBR=3,1,"PWD","tim"

OK
AT+CGDCONT=1,"IP","timbrasil.br"

OK
AT+CGACT=1,1

OK
AT+SAPBR=1,1

OK
AT+SAPBR=2,1

+SAPBR: 1,1,"100.88.137.111"

OK
AT+CGATT=1

OK
AT+CIPMUX=1

OK
AT+CIPQSEND=1

OK
AT+CIPRXGET=1

OK
AT+CSTT="timbrasil.br","tim","tim"

OK
AT+CIICR

OK
AT+CIFSR;E0

100.88.137.111

OK
AT+CDNSCFG="8.8.8.8","8.8.4.4"

OK
 success
AT+CGATT?

+CGATT: 1

OK
AT+CIFSR;E0

100.88.137.111

OK
GPRS connected
Performing HTTPS GET request... AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,,"","","","INITIAL"

OK
AT+CIPCLOSE=0,1

+CME ERROR: operation not allowed
AT+CIPSSL=1
AT+SSLOPT=0,1

OK
AT+CIPSTART=0,"TCP","vsh.pp.ua",443

OK

OK

0, CLOSE OK
failed to connect
Waiting for network...AT+CREG?

+CREG: 0,1

OK
 success
AT+CREG?

+CREG: 0,1

OK
Network connected
Connecting to timbrasil.brAT+CIPSHUT

SHUT OK
AT+CGATT=0

+SAPBR 1: DEACT

OK
AT+SAPBR=3,1,"Contype","GPRS"

OK
AT+SAPBR=3,1,"APN","timbrasil.br"

OK
AT+SAPBR=3,1,"USER","tim"

OK
AT+SAPBR=3,1,"PWD","tim"

OK
AT+CGDCONT=1,"IP","timbrasil.br"

OK
AT+CGACT=1,1

OK
AT+SAPBR=1,1

OK
AT+SAPBR=2,1

+SAPBR: 1,1,"100.121.14.6"

OK
AT+CGATT=1

OK
AT+CIPMUX=1

OK
AT+CIPQSEND=1

OK
AT+CIPRXGET=1

OK
AT+CSTT="timbrasil.br","tim","tim"

OK
AT+CIICR

OK
AT+CIFSR;E0

100.121.14.6

OK
AT+CDNSCFG="8.8.8.8","8.8.4.4"

OK
 success
AT+CGATT?

+CGATT: 1

OK
AT+CIFSR;E0

100.121.14.6

OK
GPRS connected
Performing HTTPS GET request... AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,,"","","","INITIAL"

OK
AT+CIPCLOSE=0,1

+CME ERROR: operation not allowed
AT+CIPSSL=1
AT+SSLOPT=0,1

OK
AT+CIPSTART=0,"TCP","vsh.pp.ua",443

OK

OK

0, CLOSE OK
failed to connect
```

adrianca88, 2021-08-05:
Hello,

You can find some info here:

https://github.com/vshymanskyy/TinyGSM/issues/29

Regards,
Adrián.

itisluiz, 2021-08-09:
Hi, that doesn't really illustrate my problem well, I am able to make HTTPS requests, but as I stated, those would only fail on the website provided with the HTTPS TinyGSM example file. I have provided everything I could think of on my original issue, but sadly, I'm unable to test anything further since I don't have my module anymore.

vaughanswanlund, 2022-08-16:
Hi, 
I am experiencing this issue exactly as described above.
My aim is to use the Mongodb API endpoints but I am experiencing the same issue as I do with **vsh.pp.ua**, If I can resolve why it doesn't work with vsh.pp.ua then I should be able to do the same with Mongodb.

Does anyone know what additional steps I could investigate to determine why I am experiencing this?

**I am using**: 
Modem: ATI SIMCOM SIM808 R14.18
Main processor board: Mega2560
TinyGSM version: 0.11.5
Code: examples\HttpsClient\HttpsClient.ino

**Scenario, steps to reproduce**
Attempt to establish a secure connection to the host in the example ([vsh.pp.ua](https://vsh.pp.ua/))
When I use the example sketch with the default settings:
Server: vsh.pp.ua
Resource: /TinyGSM/logo.txt

The result is:
```
AT+CIPSTART=0,"TCP","vsh.pp.ua",443
AT+CIPSTART=0,"TCP","vsh.pp.ua",443

OK

0, CLOSE OK
failed to connect
```

However, if I change the server and resource:
Server: www.google.com
Resource: /robots.txt 

The result is:
```
AT+CIPSTART=0,"TCP","www.google.com",443
AT+CIPSTART=0,"TCP","www.google.com",443

OK

0, CONNECT OK
```
Then sketch continues to get the resource successfully.

closed 2023-01-09 by itisluiz