vshymanskyy/TinyGSM #553: MQTT Broker Over GPRS with SIM800C drop connection of cliente GSM Esp32

mdonalisio, 2021-06-10:
I have a problem with this configuration, because PubSubClient Lib close connection from broker when i connect Yellow and Green, I am using TinyGSM for Gprs, over 2G, library and PubSubClient for Mqtt Client, but all work fine without using dscKeyBus Library but, when i using this library, the connection with broker is close in random time. I Only call dsc.begin() and dsc.loop(). And i using ESP32

Thanks

SRGDamia1, 2021-07-06:
I'm sorry; I don't know anything about dscKeyBus.

romoloman, 2021-08-14:
Are you sure that the pins defined for dscKeyBus Library are not conflicting with the pins used by the modem ?

SRGDamia1, 2021-11-24:
Have you been able to resolve this?  Can this issue be closed?