vshymanskyy/TinyGSM #560: Newbie question: documentation on library function

tradexsrl, 2021-07-05:
[ ] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[ ] Bug or issue with library functionality (ie, sending data over TCP/IP)
[ ] Question or request for help

### What are you working with?

Modem: 
Main processor board: 
TinyGSM version: 
Code: 

### Scenario, steps to reproduce

### Expected result

### Actual result

### Debug and AT command log

where can I find documentation on all functions for this library similarly to arduino reference?

SRGDamia1, 2021-07-06:
I'm sorry, there's some documentation in the ReadMe and Wiki here on GitHub, but there isn't a separate detailed documentation page available.

star297, 2021-12-07:
I could use some documentation on each function.
it's difficult to know which function is blocking, has a specified or adjustable timeout and what it returns without trawling through the header files and even then its not always obvious.