vshymanskyy/TinyGSM #561: MQTT mosquitto on AWS VPS refuses SIM800L connection

spara7C5, 2021-07-06:
[X ] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Bug or issue with library functionality (ie, sending data over TCP/IP)

### What are you working with?

Modem: SIM800L
Main processor board: ESP32 WROVER B, LILYGO T-CALL
TinyGSM version: see log
Code: basic mqtt example with replacing TinyGsmClient with TinyGsmClientSecure

### Scenario, steps to reproduce
properly set up a VPS instance on AWS server.
The mqtt Mosquitto broker has been tested on the VPS via PC with MQTT EXPLORER client software
You can even find IP on the log and try by yourself!

### Expected result
T-CALL board is a MQTT client which can publish and subscribe to broker

### Actual result
T-CALL cannot even establish TCP connection 

### Debug and AT command log

AT

OK
AT&W

OK
AT+CFUN=0

+CPIN: NOT READY

OK
AT+CFUN=1,1

OK
[9316] ### TinyGSM Version: 0.11.4
[9317] ### TinyGSM Compiled Module:  TinyGsmClientSIM800
AT
��AT

OK
ATE0

OK
AT+CMEE=2

OK
AT+GMM

SIMCOM_SIM800L

OK
[9629] ### Modem: SIMCOM SIM800L
[9629] ### Modem: SIMCOM SIM800L
AT+CLTS=1

OK
AT+CBATCHK=1

OK
AT+CPIN?

+CME ERROR: CFUN state is 0 or 4
AT+CPIN?

RDY

+CFUN: 1

+CPIN: READY

+CPIN: READY

OK
ATI

SIM800 R14.18

OK
Modem Info: SIM800 R14.18
AT+CPIN?

+CPIN: READY

OK
Waiting for network...AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

Call Ready

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

SMS Ready

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?  

+CREG: 0,2

OK        
AT+CREG?  

+CREG: 0,2

OK        
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,5

OK
 success
AT+CREG?

+CREG: 0,5

OK
Network connected
Connecting to TMAT+CIPSHUT

SHUT OK
AT+CGATT=0

OK
AT+SAPBR=3,1,"Contype","GPRS"

OK
AT+SAPBR=3,1,"APN","TM"

OK
AT+CGDCONT=1,"IP","TM"

OK
AT+CGACT=1,1

+CTZV: +8,1
[25912] ### Network time zone updated.

*PSUTTZ: 2021,7,6,21,53,15,"+8",1
[25916] ### Network time and time zone updated.

DST: 1
[25917] ### Daylight savings time state updated.

OK
AT+SAPBR=1,1

OK
AT+SAPBR=2,1

+SAPBR: 1,1,"10.193.30.248"

OK
AT+CGATT=1

OK
AT+CIPMUX=1

OK
AT+CIPQSEND=1

OK
AT+CIPRXGET=1

OK
AT+CSTT="TM","",""

OK
AT+CIICR

OK
AT+CIFSR;E0

10.193.30.248

OK
AT+CDNSCFG="8.8.8.8","8.8.4.4"

OK
 success
AT+CGATT?

+CGATT: 1

OK
AT+CIFSR;E0

10.193.30.248

OK
GPRS connected
AT+CREG?

+CREG: 0,5

OK
AT+CIPRXGET=4,1

+CIPRXGET: 4,1,0

OK
AT+CIPSTATUS=1

+CIPSTATUS: 1,,"","","","INITIAL"

OK
=== MQTT NOT CONNECTED ===
Connecting to 50.203.119.154AT+CIPCLOSE=1,1

+CME ERROR: operation not allowed
AT+CIPSSL=1

OK
AT+CIPSTART=1,"TCP","50.203.119.154",1883

OK
[102782] ### Unhandled: OK
 fail
AT+CREG?

1, CONNECT FAIL

THANK YOU ALL IN ADVANCE!

SRGDamia1, 2021-07-08:
Have you been able to connect to other sites, particularly with SSL?

spara7C5, 2021-07-08:
Hi/Ciao,
I performed an HTTPS GETsuccessfully

msentuerk, 2021-07-24:
I have got the same issue. Specially this line i am thinking is the problem at for:

> OK
=== MQTT NOT CONNECTED ===
Connecting to 50.203.119.154AT+CIPCLOSE=1,1

+CME ERROR: operation not allowed
AT+CIPSSL=0

AT+CIPSLL=0  must be 1 for me. Do i need to make some etra settings?

SRGDamia1, 2021-11-24:
Have you been able to resolve this?  Can this issue be closed?