vshymanskyy/TinyGSM #564: Problem connecting ESP32 to SIM7000C Waveshare NB-IOT Extension board

osugiw, 2021-07-19:
Hi everyone, currently I'm working on ESP32 and SIM7000C Waveshare NB-IOT [](https://www.waveshare.net/wiki/SIM7000C_NB-IoT_HAT). What I'm trying to do is sending AT Command from Serial monitor to SIM7000C Waveshare extension board through serial communication (RX and TX). The wiring diagram that I use look like this:

[image]

I'm using code example from TinyGSM which is **ALLFunctions**  example and changing the modem type into SIM7000 and add hardwareserial code. here's the code:

`/**************************************************************
 *
 * TinyGSM Getting Started guide:
 *   https://tiny.cc/tinygsm-readme
 *
 * NOTE:
 * Some of the functions may be unavailable for your modem.
 * Just comment them out.
 *
 **************************************************************/

// Select your modem:
//#define TINY_GSM_MODEM_SIM800
// #define TINY_GSM_MODEM_SIM808
// #define TINY_GSM_MODEM_SIM868
// #define TINY_GSM_MODEM_SIM900
 #define TINY_GSM_MODEM_SIM7000
// #define TINY_GSM_MODEM_SIM7000SSL
// #define TINY_GSM_MODEM_SIM7080
// #define TINY_GSM_MODEM_SIM5360
// #define TINY_GSM_MODEM_SIM7600
// #define TINY_GSM_MODEM_UBLOX
// #define TINY_GSM_MODEM_SARAR4
// #define TINY_GSM_MODEM_M95
// #define TINY_GSM_MODEM_BG96
// #define TINY_GSM_MODEM_A6
// #define TINY_GSM_MODEM_A7
// #define TINY_GSM_MODEM_M590
// #define TINY_GSM_MODEM_MC60
// #define TINY_GSM_MODEM_MC60E
// #define TINY_GSM_MODEM_ESP8266
// #define TINY_GSM_MODEM_XBEE
// #define TINY_GSM_MODEM_SEQUANS_MONARCH

// Set serial for debug console (to the Serial Monitor, default speed 115200)
#define SerialMon Serial

// Set serial for AT commands (to the module)
// Use Hardware Serial on Mega, Leonardo, Micro
#ifndef __AVR_ATmega328P__
#define SerialAT Serial1

// or Software Serial on Uno, Nano
#else
#include <SoftwareSerial.h>
//SoftwareSerial SerialAT(2, 3);  // RX, TX
HardwareSerial Serial1(1);
#define SerialAT Serial1
#endif

// See all AT commands, if wanted
// #define DUMP_AT_COMMANDS

// Define the serial console for debug prints, if needed
#define TINY_GSM_DEBUG SerialMon

// Range to attempt to autobaud
// NOTE:  DO NOT AUTOBAUD in production code.  Once you've established
// communication, set a fixed baud rate using modem.setBaud(#).
#define GSM_AUTOBAUD_MIN 9600
#define GSM_AUTOBAUD_MAX 57600

// Add a reception delay, if needed.
// This may be needed for a fast processor at a slow baud rate.
[... 407 lines cut ...]
    DBG("GPRS disconnect: Failed.");
  }
#endif

#if TINY_GSM_TEST_WIFI
  modem.networkDisconnect();
  DBG("WiFi disconnected");
#endif

  // Try to power-off (modem may decide to restart automatically)
  // To turn off modem completely, please use Reset/Enable pins
  modem.poweroff();
  DBG("Poweroff.");
#endif

  DBG("End of tests.");

  // Do nothing forevermore
  while (true) { modem.maintain(); }
}`

It couldn't communicate with the extension board because the results look like this:
[image]

Any advice or help would be appreciate! Thanks everyone!

osugiw, 2021-07-19:
The documentation:

https://www.waveshare.net/wiki/SIM7000C_NB-IoT_HAT

closed 2021-09-09 by osugiw

Koivader, 2023-03-19:
@osugiw How did you solve the problem. I have a similar problem.

osugiw, 2023-03-19:
> @osugiw How did you solve the problem. I have a similar problem.

Sorry, I little forget how I solved this problem, If I'm not wrong my problem that time because of the baudrate of UART

osugiw, 2023-03-20:
> @osugiw How did you solve the problem. I have a similar problem.

I have just remembered, my problem that time was the sim card provider isn't compatible with SIM7000. You have to look up the datasheet for the bandwidth frequencies that supported in SIM7000.

Koivader, 2023-03-20:
Thank you for your quick answer. Unfortunately this wasn't my problem.

In case someone has a similar issue than I have, for me it was, that I used the wrong HardwareSerial Pins.

`#define ESP32_TX 16  // ESP32 hardware serial RX2 (GPIO16)
#define ESP32_RX 17  // ESP32 hardware serial TX2 (GPIO17)
// For ESP32 hardware serial
#include <HardwareSerial.h>
HardwareSerial SerialAT(1);`

and in 'setup()':
`SerialAT.begin(115200, SERIAL_8N1, ESP32_RX,ESP32_TX);  // baud rate, protocol, ESP32 RX pin, ESP32 TX pin`

aleemzzz, 2023-12-29:
> Thank you for your quick answer. Unfortunately this wasn't my problem.
> 
> In case someone has a similar issue than I have, for me it was, that I used the wrong HardwareSerial Pins.
> 
> ```c
> #define ESP32_TX 16  // ESP32 hardware serial RX2 (GPIO16)
> #define ESP32_RX 17  // ESP32 hardware serial TX2 (GPIO17)
> // For ESP32 hardware serial
> #include <HardwareSerial.h>
> HardwareSerial SerialAT(1);
> ```
> 
> and in `setup()`: `SerialAT.begin(115200, SERIAL_8N1, ESP32_RX,ESP32_TX); // baud rate, protocol, ESP32 RX pin, ESP32 TX pin`
> 
> Edit: Changed Code Format

 
Hi , there can you provide me the code of your project?

Koivader, 2024-01-04:
Hi, I was starting with this example https://github.com/vshymanskyy/TinyGSM/blob/master/examples/MqttClient/MqttClient.ino.

Here is my complete code, although I'm not too sure if it was working. 😅

```cpp
/**************************************************************
 *
 * For this example, you need to install PubSubClient library:
 *   https://github.com/knolleary/pubsubclient
 *   or from http://librarymanager/all#PubSubClient
 *
 * TinyGSM Getting Started guide:
 *   https://tiny.cc/tinygsm-readme
 *
 * For more MQTT examples, see PubSubClient library
 *
 **************************************************************
 * This example connects to HiveMQ's showcase broker.
 *
 * You can quickly test sending and receiving messages from the HiveMQ webclient
 * available at http://www.hivemq.com/demos/websocket-client/.
 *
 * Subscribe to the topic GsmClientTest/ledStatus
 * Publish "toggle" to the topic GsmClientTest/led and the LED on your board
 * should toggle and you should see a new message published to
 * GsmClientTest/ledStatus with the newest LED status.
 *
 **************************************************************/

// Select your modem:
// #define TINY_GSM_MODEM_SIM800
// #define TINY_GSM_MODEM_SIM808
// #define TINY_GSM_MODEM_SIM868
// #define TINY_GSM_MODEM_SIM900
#define TINY_GSM_MODEM_SIM7000
// #define TINY_GSM_MODEM_SIM7000SSL
// #define TINY_GSM_MODEM_SIM7080
// #define TINY_GSM_MODEM_SIM5360
// #define TINY_GSM_MODEM_SIM7600
// #define TINY_GSM_MODEM_UBLOX
// #define TINY_GSM_MODEM_SARAR4
// #define TINY_GSM_MODEM_M95
// #define TINY_GSM_MODEM_BG96
// #define TINY_GSM_MODEM_A6
// #define TINY_GSM_MODEM_A7
// #define TINY_GSM_MODEM_M590
// #define TINY_GSM_MODEM_MC60
// #define TINY_GSM_MODEM_MC60E
// #define TINY_GSM_MODEM_ESP8266
// #define TINY_GSM_MODEM_XBEE
// #define TINY_GSM_MODEM_SEQUANS_MONARCH

#include "mySim7000.h"
#include "mySim7x00.h"

// Set serial for debug console (to the Serial Monitor, default speed 115200)
#define SerialMon Serial

#define ESP32_TX 16  // ESP32 hardware serial RX2 (GPIO16)
#define ESP32_RX 17  // ESP32 hardware serial TX2 (GPIO17)
// For ESP32 hardware serial
#include <HardwareSerial.h>
// HardwareSerial SerialAT(1);

/*
// Set serial for AT commands (to the module)
// Use Hardware Serial on Mega, Leonardo, Micro
#ifndef __AVR_ATmega328P__
#define SerialAT Serial1

[... 152 lines cut ...]
    modem.init();
    */
}

void loop() {
    // if ((loop_counter + 10) % 100 == 0) {
    //     SerialMon.print("Signal quality:");
    //     SerialMon.println(modem.getSignalQuality());
    //     modem.sendAT(GF("+CSQ"));
    //     SerialMon.print("Signal quality CMD send: ");
    //     SerialMon.print(modem.waitResponse(GF("+CSQ:")));
    //     SerialMon.println(modem.stream.readStringUntil('\n'));

    //     SerialMon.print("Is network connected: ");
    //     SerialMon.println(modem.getLocalIP());
    //     SerialMon.println(modem.isNetworkConnected());
    // }
    // if (loop_counter == 100) {
    //     SerialMon.println("sending UDP Message");
    // }

    delay(300);
    loop_counter++;
}
```