vshymanskyy/TinyGSM #567: Multiple clients with TINY_GSM_MODEM_SIM7000SSL fails on http connection for file download

Ollebolle, 2021-07-21:
[x] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[x] Bug or issue with library functionality (ie, sending data over TCP/IP)
[ ] Question or request for help

### What are you working with?

Modem: SIM7000
Main processor board: TTGO ESP32
TinyGSM version: 0.11.4
Code: 

```C
// Set serial for debug console (to the Serial Monitor, default speed 115200)
#define SerialMon Serial

// Set serial for AT commands (to the module)
// Use Hardware Serial on Mega, Leonardo, Micro
#define SerialAT Serial1
#define TINY_GSM_MODEM_SIM7000SSL
#define TINY_GSM_RX_BUFFER 512 // Set RX buffer to 1Kb
#define SerialAT Serial1

// See all AT commands, if wanted
#define DUMP_AT_COMMANDS

// set GSM PIN, if any
#define GSM_PIN ""

// Your GPRS credentials, if any
const char apn[] = ""; //SET TO YOUR APN
const char gprsUser[] = "";
const char gprsPass[] = "";

#include <TinyGsmClient.h>
#include <SPI.h>
#include <SD.h>
#include <Ticker.h>
#include <PubSubClient.h>
#include <Preferences.h>
#include "GSMFOTAUpdate.h" 

#ifdef DUMP_AT_COMMANDS
#include <StreamDebugger.h>
StreamDebugger debugger(SerialAT, SerialMon);
TinyGsm modem(debugger);
#else
TinyGsm modem(SerialAT);
#endif

#define uS_TO_S_FACTOR 1000000ULL // Conversion factor for micro seconds to seconds
#define TIME_TO_SLEEP 60          // Time ESP32 will go to sleep (in seconds)

#define UART_BAUD 9600
#define PIN_DTR 25
#define PIN_TX 27
#define PIN_RX 26
#define PWR_PIN 4

#define SD_MISO 2
#define SD_MOSI 15
#define SD_SCLK 14
#define SD_CS 13
#define LED_PIN 12
#define PIN_BAT 35 /* read battery voltage, Disabled when using USB - ONLY INPUT PIN */
[... 335 lines cut ...]

### Expected result
I'm expecting both `TinyGsmClientSecure` and `TinyGsmClient` to work together. At 

### Actual result
`clientOta` fails to even connect to the server.

### Debug and AT command log

```
♦ ♦ UPDATE AVALIBLE ♦ ♦

Connecting to: *.amazonaws.com

OK
AT+CACLOSE=1

ERROR
AT+CACID=1

OK
AT+CASSLCFG=1,ssl,0

OK
AT+CAOPEN=1,"*.amazonaws.com",80

SRGDamia1, 2021-11-24:
I'm combing through months of old issues.  Were you able to resolve this?

pitlochry, 2022-07-21:
I have the same issue: How do you combine `TinyGsmClientSecure` and `TinyGsmClient` on SIM7000 to work together? Readme says: "TCP and SSL connections can usually be mixed up to the total number of possible connections" But how do you do it? 

@Ollebolle Did you succeed with TinyGSM or did you use something different?

gotnull, 2023-01-21:
Anyone get an update on this?