vshymanskyy/TinyGSM #572: GPRS connection CLOSED before entire file has been received

FordJ, 2021-07-28:
[x] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
[ ] Bug or problem compiling the library
[x] Bug or issue with library functionality (ie, sending data over TCP/IP)
[ ] Question or request for help

### What are you working with?

Modem: SIM868
Main processor board: SAMD51
TinyGSM version: Latest, but also several old versions

### Scenario
Trying to download a file over GPRS, everything working as expected, but the connection is CLOSED before the entire file has been received by my code.
It's been tried without SSL, with SSL, different servers, different files, on 5+ boards with different revisions of SIM868, different SIM cards, etc.
The exact same number of chars are downloaded when running identical code again, and when downloading a larger file, more data is downloaded, but doesn't miss the same number of chars.

E.g. 9472 byte file downloads 8416, 4736 byte file downloads 3296

I've experimented with different buffer sizes (both `TINY_GSM_RX_BUFFER` and RingBuffer, and get different read lengths, but still behaving in the same way.

Any suggestions on causes or further debugging?

Code: 
`#define SerialMon SERIAL_PORT_MONITOR

#define OTA_PACKET_SIZE 128
#define TINY_GSM_RX_BUFFER (4 * OTA_PACKET_SIZE + 1)

#define TINY_GSM_MODEM_SIM868
#define TINY_GSM_YIELD() {delay(1);}
#define TINY_GSM_DEBUG	SERIAL_PORT_USBVIRTUAL

#include <TinyGsmClient.h>

TinyGsm modem(MODEM_SERIAL);
TinyGsmClientSecure client(modem);

void setup(){
	SerialMon.begin(115200);
	while(!SERIAL_PORT_MONITOR);
	SerialMon.println("Serial connected");

	MODEM_SERIAL.begin(115200);
	MODEM_SERIAL.setTimeout(1);

	pinMode(PIN_MODEM_WAKE, OUTPUT);
	digitalWrite(PIN_MODEM_WAKE, !LEVEL_MODEM_WAKE);
	delay(5000);

	MODEM_SERIAL.println("AT");
	delay(100);
	MODEM_SERIAL.println("AT");
	delay(100);
	MODEM_SERIAL.println("AT+CSCLK=0");

	modem.restart();

	modem.init();

	SerialMon.print("Modem Info: ");
	SerialMon.println(modem.getModemInfo());
}

void loop(){
	uint32_t OTAstartMillis = millis();
	SerialMon.println("Waiting for network...");
[... 384 lines cut ...]
ABCDEFGHIJKLMNOPQ

[70467] ### AT: +CIPRXGET=4, 0

[70492] ### AT: +CIPSTATUS= 0

(GPRS:239:loop) >>  Head length = 288

(GPRS:240:loop) >> Start length = 9472

(GPRS:241:loop) >>  Read length = 8416

(GPRS:242:loop) >>  Update took = 59384ms

[70565] ### AT: +CIPCLOSE= 0 ,1

Server disconnected

[70582] ### AT: +CIPSHUT

[70597] ### AT: +CGATT=0

GPRS disconnected

Done`

SRGDamia1, 2021-11-24:
Have you been able to resolve this?  Can this issue be closed?

lfdominguez-servinfo, 2023-04-21:
This happen to me and the only way that i can fix it is changing the Serial port connection speed, from 115200 to something low low, like 9600.

furqon2710, 2023-07-21:
I am having the same problem.  it disconnect on 95 % progress twice when i am using 9600 baudrate. after that i increasing the baudrate into 115200 and it disconnect on  85%. Also i dont know why it take a long time to download it,  for getting that 95%  take an hour.