vshymanskyy/TinyGSM #575: lilygo ttgo sim7600 and ssl

tradexsrl, 2021-08-03:
Hi , just received this board , i need to connect with ssl but i don't find in the library a .h file for this modem with support of ssl.
please let me know
thank's

A-n-d-y-K, 2021-08-18:
Hello @tradexsrl ,
I´m in the same situation with my SIM7600E. Have you already managed to use SSL with this module?
If yes could you share your way of doing?
Thanks in advance

tradexsrl, 2021-08-19:
Hi , no at the moment i'm testing with sim7000 because sim7600e have also problem with gps and I need to understand quickly if esp32+simcom it will be the right choice for my projects

tradexsrl, 2021-08-19:
i have found this repository: https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7600X/blob/master/examples/SSL_server/SSL_server.ino
which contains an ssl example. i saw they are using some AT command , so i suspect tinygsm do not support ssl for this modem at the moment....i think we need to read the simcom AT command manual because in my case i need to connect to a mqtt server in ssl...which will be not so simple without a function in the library....

A-n-d-y-K, 2021-08-20:
Hello @tradexsrl ,
thanks for your reply. I already know the example you have linked. Sadly it is for an server application. I want to use the `SIM7600` in the role client.
Meanwhile I´ll succeed. Thanks to the link provided by @Kiddtang in this comment: https://github.com/vshymanskyy/TinyGSM/issues/535#issuecomment-869117435.
So far I´m using an pre generated certificate stored in flash memory of ESP32. The goal is to the `SIM7600`.
Is there anyone who is familar with instructing the SIM7600 Module via AT commands to download a legal cert?

Regarding your MQTT(s) issue:
On the Homepage of www.simcom.com There is an "MQTT(S)_Application Note". Maybe this helps you.
Cheers