vshymanskyy/TinyGSM #580: Issues with UBLOX SARA-G350

FStefanni, 2021-08-24:
[x ] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[x] Bug or issue with library functionality (ie, sending data over TCP/IP)
[x] Question or request for help

### What are you working with?

Modem: UBLOX SARA-G350-02S-01 (board Microe GSM 4 click)
Main processor board: ESP32
TinyGSM version: 0.10.9

### Description

Hi,

I am having quite trouble by using UBLOX SARA-G350. Basically, it does not work at all.
I have taken the HTTP client example sketch, but I am viewing the following issues:
1. The initialization prints sometimes the correct model (### Modem: u-blox SARA-G350), most often the generic string in case of error  (### Modem: u-blox Cellular Modem)
2. When the sketch tries to perform `gprsConnect()` almost always fails.
3. In the rare cases when `gprsConnect()` succeed, the HTTP request fails.

According with the chip manual, the modem reset should be hold down for **at least** 5ms, during initialization.
Actually, I have tested various delays, but non of them works...

Does anyone have my same issues?
Can someone please confirm that UBLOX SARA-G350 works fine?

Thanks,
regards.

SRGDamia1, 2021-11-24:
Were you ever able to resolve this?  I know some other firmwares of u-blox modems have been nearly-non-functional.  See #296

FStefanni, 2021-11-25:
Hi,

thank you for letting me know about the other similar issue.
No, I was not able to let it work, and now I am working with another modem.
Maybe (hopefully) in January I will be able to test it again (right now I am full on other stuff).

Regards.