vshymanskyy/TinyGSM #583: SSL MQTT Connection Fails

pablofr918, 2021-09-01:
[ x] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[x ] Bug or issue with library functionality (ie, sending data over TCP/IP)
[ ] Question or request for help

### What are you working with?

Modem: SIM800C
Main processor board: ESP32
TinyGSM version: latest
Code: 

### Scenario, steps to reproduce
Hello! i'm trying to make a SSL connection to a broker with the example given, but I can't make it possible. I have first tried to make this connection through the AT commands, and it seems to work, so I then tried to make it with the library, but connection fails everytime. The commands I use to make the connection through AT are the following:

[image]

Modifying the APN and the broker url to connect.

Now I will paste the code I'm using to try the connection:

```
/**************************************************************
 *
 * To run this tool you need StreamDebugger library:
 *   https://github.com/vshymanskyy/StreamDebugger
 *   or from http://librarymanager/all#StreamDebugger
 *
 * TinyGSM Getting Started guide:
 *   https://tiny.cc/tinygsm-readme
 *
 **************************************************************/

// Select your modem:
#define TINY_GSM_MODEM_SIM800
// #define TINY_GSM_MODEM_SIM808
// #define TINY_GSM_MODEM_SIM868
// #define TINY_GSM_MODEM_SIM900
// #define TINY_GSM_MODEM_SIM7000
// #define TINY_GSM_MODEM_SIM7000SSL
// #define TINY_GSM_MODEM_SIM7080
// #define TINY_GSM_MODEM_SIM5360
// #define TINY_GSM_MODEM_SIM7600
// #define TINY_GSM_MODEM_UBLOX
// #define TINY_GSM_MODEM_SARAR4
// #define TINY_GSM_MODEM_M95
// #define TINY_GSM_MODEM_BG96
// #define TINY_GSM_MODEM_A6
// #define TINY_GSM_MODEM_A7
// #define TINY_GSM_MODEM_M590
// #define TINY_GSM_MODEM_MC60
// #define TINY_GSM_MODEM_MC60E
// #define TINY_GSM_MODEM_ESP8266
// #define TINY_GSM_MODEM_XBEE
// #define TINY_GSM_MODEM_SEQUANS_MONARCH

// Set serial for debug console (to the Serial Monitor, default speed 115200)
#define SerialMon Serial

// Set serial for AT commands (to the module)
#define SerialAT Serial2

// Increase RX buffer to capture the entire response
// Chips without internal buffering (A6/A7, ESP8266, M590)
// need enough space in the buffer for the entire response
[... 248 lines cut ...]
    }
    delay(100);
    return;
  }

  mqtt.loop();
}
```
I have added this part:
```
 modem.sendAT(GF("+CIPSSL=1"));
  int rsp1 = modem.waitResponse();
  if (rsp1 != 1)
  {
    Serial.printf("modem +CIPSSL=1 failed");
  }
  else{
    Serial.print("SSL activated");
  }
```
Just to make exactly the same steps than the AT commands. I turn SSL option on, and then the mqtt connects to the broker, port given. But it fails.

Someone can help me, or has just made a connection to a broker with SSL?

Thank you very much!

RupeshPilania, 2021-09-17:
@pablofr918 I have few queries here.
1. Are you using self signed certificate?
2. Have you uploaded the certificates to module?
I was using SSLClient library with TinyGSM library with ESP32 and SIM800L module with some fixes instead of using internal ssl support given by GSM module. I am able to get stable mqtt connection(tested for 5 hours) with AWS IoT Core.  Sometimes ssl error comes with SSLClient library also. It might be due to unstable network. 
I will explore internal SSL support provided by the module.

roysG, 2021-10-31:
Hi @pablofr918 ,
I am here in the same issue, can you please upload your code or some example?
Thanks!

roysG, 2021-12-04:
?

roysG, 2021-12-04:
@pablofr918

icabrera29, 2023-04-14:
@pablofr918

SherifMEldeeb, 2024-09-04:
Greetings @pablofr918   @RupeshPilania,
I am having same issues with same architecture how ever in my case the connection is very unstable I get a lot of unexplained ssl errors. I'm using a let's encrypt certificate could you give me any clue to get it working.
Thx in advance.