vshymanskyy/TinyGSM #584: Force operator

edge7, 2021-09-01:
[ ] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?
 Question or request for help

### What are you working with?

Modem: 
Main processor board: 
TinyGSM version: 

### Scenario, steps to reproduce
Hi, am using a sim that has a virtual network, so it relies on some physical operators. Do you know how can I try to force the registration with a particular one? (Vodafone).

edge7, 2021-09-02:
I solved with:
("AT+COPS=1,2,\"22210\"")
where 22210 is the concatenation of the operator MCC and MNC to which one wants to connect.

closed 2021-09-02 by edge7

ejri, 2021-12-27:
Would you please provide more information on this?

Where/when do you call this? Is there a way to also include information about the band?

E.g. MCC=302, MNC=610, band=5

Would you please share a code snippet?

edge7, 2022-01-24:
I use it to get it connected.
Full example:
`bool connectToGPRS(){
  

  #ifdef TINY_GSM_MODEM_SIM7000
    modem.setNetworkMode(38);
    delay(100);
    modem.setPreferredMode(2);
    SerialAT.println("AT+COPS=1,2,\"22210\"");     //Get connection type and band
    delay(5500);
    if (SerialAT.available()) {
      Serial.println("Response");
      String r = SerialAT.readString();
      Serial.println(r);
    } else{
      Serial.println("No response");
    }
  #endif
  
  
  bool toReturn = false;
  for( char i=0; i < 5; i++){
    if (!modem.waitForNetwork()) {
      SerialMon.println(" failed to Wait For Netowrk");
      
      delay(10000);
      continue;
    }
    SerialMon.println(" success to connect to Netowrk");
  
    if (modem.isNetworkConnected()) { SerialMon.println("Network connected"); 
    
    } else continue;
    
    bool res=true;
  do {
    res = modem.setNetworkMode(38);
    Serial.println(res);
    delay(500);
  } while (!res);
  Serial.println("OK LTE");
  do {
    res = modem.setPreferredMode(2);
    Serial.println(res);
    delay(500);
  } while (!res);
  Serial.println("OK NBIOT");
  SerialMon.print(("Connecting to "));
  SerialMon.print(apn);
  if (!modem.gprsConnect(apn, gprsUser, gprsPass)) {
    SerialMon.println(" fail to connect to APN");
    delay(10000);
    continue;
  }

  SerialMon.println(" success to connect to APN");

  
  if (modem.isGprsConnected()) { 
    SerialMon.println("GPRS connected"); toReturn= true;

  break;}
  else{
    Serial.println("Not connected");
  }
  }
  String ccid = modem.getSimCCID();
  Serial.println("CCID: " + ccid);

  String imei = modem.getIMEI();
  Serial.println("IMEI: " + imei);

  String cop = modem.getOperator();
  Serial.println("Operator: " + cop);

  IPAddress local = modem.localIP();
  Serial.println("Local IP: " + String(local));

  int csq = modem.getSignalQuality();
  Serial.println("Signal quality: " + String(csq));

  #ifdef TINY_GSM_MODEM_SIM7000
  SerialAT.println("AT+CPSI?");     //Get connection type and band
  delay(500);
  if (SerialAT.available()) {
    String r = SerialAT.readString();
    Serial.println(r);
  }
  #endif
  
  return toReturn;
}`