vshymanskyy/TinyGSM #586: Location not accurate, but nearby

sixtybi, 2021-09-12:
[X] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[X] Bug or issue with library functionality (ie, sending data over TCP/IP)
[ ] Question or request for help

### What are you working with?

Modem: SIM800L R14.18

Main processor board: ESP32
TinyGSM version: 0.11.3
Code: modem.getGsmLocation(&lat, &lon, &accuracy, &year, &month, &day, &hour, &min, &sec)

My problem is that the received location is, however nearby, very inaccurate.
Few km away in most cases.
I tried to use 3 cell location method, but I haven't received any location using that method.

### Debug and AT command log
AT+CIPSTATUS=0

+CIPSTATUS: 0,0,"TCP","139.59.206.133","80","CONNECTED"

OK
Checking location
AT+CLBS=4,1

+CLBS: 0,18.616992,47.235787,550,21/09/12,17:23:26

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,0,"TCP","139.59.206.133","80","CONNECTED"

OK
AT+CIPSEND=0,21

Humancell, 2021-09-30:
This has nothing to do with the library.  It is what is being returned by the modem, from the carrier.  GSM Location by tower triangulation is really going to vary in accuracy based on your location.

SRGDamia1, 2021-11-24:
@sixtybi - Can I close this issue?

sixtybi, 2021-11-24:
Yes, thank you.

closed 2021-11-24 by SRGDamia1