vshymanskyy/TinyGSM #591: Latitude and Longitude switched?

tilchl, 2021-09-30:
[X] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[X] Bug or issue with library functionality (ie, sending data over TCP/IP)
[X] Question or request for help

### What are you working with?

Modem: SIM7000
Main processor board: ESP32
TinyGSM version: 0.11.4
Code: 
```c#
modem.getGsmLocation(&gsmLat, &gsmLon);
```
### Expected result
```c#
Serial.println(gsmLat,8) // 52.xxxxxxxx
Serial.println(gsmLon,8) // 9.xxxxxxxx
```
Position in Germany

### Actual result
```c#
Serial.println(gsmLat,8) // 9.xxxxxxxx
Serial.println(gsmLon,8) // 52.xxxxxxxx
```
Position in the Indian Ocean
For some reason the latitude and longitude are switched.

adrianca88, 2021-10-20:
Hello,

It is strange, according to the documentation the implementation is correct:

https://github.com/vshymanskyy/TinyGSM/blob/09e679e9433e09bf6d2af51121074c82cd4c4507/src/TinyGsmClientSIM70xx.h#L341

http://www.mt-system.ru/sites/default/files/documents/sim7000_series_at_command_manual_v1.06.pdf (page 224)

Could you enable DUMP_AT_COMMANDS and paste here the result?

SRGDamia1, 2021-11-24:
Any update on this?

tilchl, 2021-11-24:
Seems that some cell towers return them switched and some do not. No real way of fixing this i belive

closed 2021-11-24 by tilchl

SRGDamia1, 2021-11-24:
Sorry!  Thank you for closing this.