vshymanskyy/TinyGSM #592: SIM7080G not registering, except after accidentally treating as SIM7000G?

Humancell, 2021-09-30:
[X] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[ ] Bug or issue with library functionality (ie, sending data over TCP/IP)
[X] Question or request for help

### What are you working with?

Modem: SIM7080G - R1951.01
Main processor board: Raspberry Pi Pico
TinyGSM version: 0.11.4
Code: All Functions (enhanced)

### Scenario, steps to reproduce
I'm just trying to get the All Functions example to work.

### Expected result
I expected it would work.

### Actual result
At first, I was unable to get the modem to work at all, but I added a small bit of code to power cycle the modem in setup():

  // reset the modem
  DBG("Hard reset modem ...");
  digitalWrite(14, LOW);
  delay(1000);
  digitalWrite(14, HIGH);
  delay(1000);
  DBG("Done!");

After this, I was get the modem communicating, however it failed to register properly.  It sticks in a loop of:

AT+CSQ
+CSQ: 99,99
OK
AT+CEREG?
+CEREG: 0,2
OK
AT+CGREG?
+CGREG: 0,2
OK

I then added some code to setup() from my older SIM7000G code (which works with the SIM7000G) to ensure that the modem was properly configured for Cat-M:

  /*
  2 Automatic
  13 GSM only
  38 LTE only
  51 GSM and LTE only
  * * * */
  bool res;
  DBG("Set Network Mode == 51 GSM and LTE only ...");
  do {
      res = modem.setNetworkMode(51);
      delay(500);
  } while (res != true);
  DBG("Done!");

  /*
  1 CAT-M
  2 NB-Iot
  3 CAT-M and NB-IoT
  * * */
  DBG("Set Preferred mode == Cat-M ...");
  do {
      res = modem.setPreferredMode(1);
      delay(500);
  } while (res != true);
  DBG("Done!");

This still didn't work.

I then ran an older SIM7000G debugger app - just a completely stripped down version of All Functions - but forgot to change the #define from SIM7000G to SIM7080G.  Running this failed.  And that is when I realized my error, and I then went back to running my All Functions configured for the SIM7080G ... and now it worked!

On this device it now works all of the time.  So it appears that SOMETHING in the SIM7000G restart or gprsConnect is doing something different from the SIM7080G flow?

I took a brand new device, and again tried with the SIM7080G code and it continues to fail the same way.  The one that was run with the SIM7000G #define once now continues to work with all of the examples.  Some setting appears to have been toggled, but I can't figure out which one.

### Debug and AT command log
[6733] Hard reset modem ...
[8653] Done!
[8653] Trying baud rate 57600 ...
[9665] Modem responded at rate 57600
[9665] Set Network Mode == 51 GSM and LTE only ...
AT+CNMP=51
AT+CNMP=51

OK
[10159] Done!
[10159] Set Preferred mode == Cat-M ...
AT+CMNB=1
AT+CMNB=1

OK
[10652] Done!
[10653] Initializing modem...
ATE0
ATE0

OK
AT+CFUN=0

+CPIN: NOT READY

OK
AT+CFUN=1,1

OK
   [41074] ### TinyGSM Version: 0.11.4
[41074] ### TinyGSM Compiled Module:  TinyGsmClientSIM7080
AT
AT
AT
AT
AT
AT
AT
AT

OK
ATE0
ATE0

OK
AT+CMEE=2

OK
AT+GMM

SIMCOM_SIM7080

OK
[42890] ### Modem: SIMCOM SIM7080
AT+CLTS=1

OK
AT+CBATCHK=1

OK
AT+CPIN?

+CPIN: READY

OK
AT+GMM

SIMCOM_SIM7080

OK
[42939] Modem Name: SIMCOM SIM7080
ATI

R1951.01

OK
[42954] Modem Info: R1951.01
AT+CPIN?

+CPIN: READY

OK
AT+CNACT=0,0

+CME ERROR: operation failed
AT+CGDCONT=1,"IP","super"

OK
AT+CGATT=1

ERROR
[43103] Waiting for network...
AT+CSQ

+CSQ: 99,99

OK
AT+CEREG?

+CEREG: 0,6

OK
AT+CGREG?

+CGREG: 0,6

OK
AT+CSQ

+CFUN: 1

+CPIN: READY

+CSQ: 99,99

OK
AT+CEREG?

+CEREG: 0,2

OK
AT+CGREG?

+CGREG: 0,2

OK
AT+CSQ

+CSQ: 99,99

OK
AT+CEREG?

+CEREG: 0,2

OK
AT+CGREG?

+CGREG: 0,2

OK

-----------------------------------------------------------------------

Any thoughts on how to further debug why this might be occurring?

Humancell, 2021-09-30:
I also went back to my first module (which is now working) and tried the ATZ and AT&F0 reset commands ... and rebooted with AT+CFUN=1,1 ... but can no longer get it back to the original failure mode.

My second module - where I have never run code using the #define for the 7000G - continues to not work.

I'm really struggling to figure out exactly which command was sent by the library that "enabled" the 7080G to work.

SRGDamia1, 2021-11-24:
Were you able to resolve this?

dlyckelid, 2021-12-06:
I am currently using 7080G without issues in TinyGSM. However in the beginning I had trouble getting it to work. Like you I tested alot of things and suddenly it worked but I now have a code that works all the time for me.

`SerialAT.begin(57600);

this->RestartModem();`

`void GSMConnection::RestartModem()
{

	this->connected = false;

	if (!modem.testAT())
	{
		DebugService::Information("initializeGSM", "Modem not started ");
		digitalWrite(4, HIGH);
		delay(1000);
		digitalWrite(4, LOW);
		delay(5000);
	}

	this->modem.restart();
	delay(5000);
	String name = this->modem.getModemName();
	DebugService::Information("GSMConnection::loop Modem Name:", name);
	String modemInfo = modem.getModemInfo();
	DebugService::Information("initializeGSM", "Modem Info: " + modemInfo);

	if (ConfigurationService::simPin && modem.getSimStatus() != 3)
	{ 
		DebugService::Information("initializeGSM", "Unlocking sim");
		modem.simUnlock(ConfigurationService::simPin.c_str()); 
	}

	this->modem.setNetworkMode(2);

	this->modem.setPreferredMode(1);

	String simCardCCID = this->modem.getSimCCID();

	if (simCardCCID == "ERROR")
	{
		noSimCard = true;
	}
	else
	{
		modem.waitForNetwork(10000, true);
	}
	DebugService::Information("initializeGSM", "Sim card CCID: " + simCardCCID);
}`

qt1, 2022-06-08:
@dlyckelid Hi, can you post log of AT commands?

rbev, 2022-11-11:
Did you ever figure this out?

I suspect I've run into this same issue - 

At one point I got all working & connecting, pushed my code to the server and did something else. After a power cycle of the modem it no longer connects. 
I suspect that I issued an AT command that made it work, but i've been unable to work out what it was.

As you can see from the log below it can see my network (telstra) and has good signal, but it sits in searching forever.

```
AT+CSQ

+CSQ: 29,99

OK
AT+CMEE=2

OK
AT+COPS=?

+COPS: (3,"Optus AU","Optus","50502",9),(3,"vodafone AU","voda AU","50503",9),(1,"Telstra Mobile","Telstra","50501",7),(1,"Telstra Mobile","Telstra","50501",9),,(0,1,2,3,4),(0,1,2)

OK
AT+CEREG?

+CEREG: 0,2

OK
AT+CGREG?

+CGREG: 0,2

OK
```

Humancell, 2022-11-11:
I have never been able to test the suggestion from above.  @sharask indicated that the band selection was set by default to a very narrow set (or one?) band.  I really need to try this, but have just not had the time.

Seeing this reminded me that maybe next week I need to dust off my Pico and SIM7080G and give it another go.

I meant to get the TinyGSM docs and see if there was a library call to make for setting the active bands in the modem, or if this was simply going to be a AT Command that has to be issued.

charles-plante, 2022-12-21:
Have you found something ? I am facing the same issue. My 7080g is not always registering, sometime a reset the power and it work and sometimes it loop forever looking for registration.

Olivier-zitoun, 2023-01-11:
Hello everybody,

I have the same problem as you, my SIM7080G was working before, and now I get the CME message 'no network service', while the SIM card is good, the AT+CSQ level is good.

If I try to force manually the connection to the network AT+COPS=1,2,"20801" command, I get an error. message..

AT+CPSI returns 'No Service, On line' ...

This is just crazy behavior !

I also tried to modify the IMEI manually (in the case my modem is black-listed by the operator),but still no connection.

dlyckelid, 2023-01-12:
Hi!

I am not sure of exactly what issue you are having but I am using 7080G in my devices and currently have more than 250 out in the world working.
I can share my code that handles the connection and initiation of the modem and hopefully that can help you find the issue you are having.

`#ifndef SerialAT
#define TINY_GSM_MODEM_SIM7080
#define SerialAT Serial2
#define GSM_AUTOBAUD_MIN 9600
#define GSM_AUTOBAUD_MAX 115200
#endif // !SerialAT

#include "GSMConnection.h"

void GSMConnection::initialize(bool active)
{
	//uint32_t modemSpeed = TinyGsmAutoBaud(SerialAT, GSM_AUTOBAUD_MIN, GSM_AUTOBAUD_MAX);
	
	//DebugService::Information("initializeGSM", "Initializing modem... ");

	this->active = active;
}

void GSMConnection::RestartModem()
{
	this->connected = false;

	DebugService::Information("initializeGSM", "Test AT ");
	if (!modem.testAT())
	{
		DebugService::Information("initializeGSM", "Modem not started ");
		if (ConfigurationService::newHardware)
		{
			DebugService::Information("initializeGSM", "Send start ON ");
			IOStatus::Sim_Power = true;
			delay(1000);
			DebugService::Information("initializeGSM", "Send start OFF ");
			IOStatus::Sim_Power = false;
		}
		else
		{
			digitalWrite(4, HIGH);
			delay(1000);
			digitalWrite(4, LOW);
		}
		delay(5000);
	}

	DebugService::Information("initializeGSM", "Restart modem ");
	this->modem.restart();
	delay(10000);
	String name = this->modem.getModemName();
	DebugService::Information("GSMConnection::loop Modem Name:", name);
	String modemInfo = modem.getModemInfo();
	DebugService::Information("initializeGSM", "Modem Info: " + modemInfo);

	if (ConfigurationService::simPin && modem.getSimStatus() != 3)
	{ 
		DebugService::Information("initializeGSM", "Unlocking sim");
		modem.simUnlock(ConfigurationService::simPin.c_str()); 
	}

	this->modem.setNetworkMode(2);

	this->modem.setPreferredMode(1);

	String simCardCCID = this->modem.getSimCCID();

	if (simCardCCID == "ERROR")
	{
		noSimCard = true;
	}
	else
	{
		noSimCard = false;
		modem.waitForNetwork(10000, true);
	}
	DebugService::Information("initializeGSM", "Sim card CCID: " + simCardCCID);
}

void GSMConnection::resetConnection()
{
	if(this->active)
	this->RestartModem();
}

String GSMConnection::getConnectionDetail()
{
	return "M " + String(modem.getSignalQuality());
}

void GSMConnection::loop(String apnAdress, String apnUser, String apnPassword)
{
	if (active)
	{
		if (this->firstRun == false)
		{
			DebugService::Information("initializeGSM", "Initializing modem... ");
			SerialAT.begin(115200, SERIAL_8N1, 22, 5);
			DebugService::Information("initializeGSM", "Serial started ");
			this->RestartModem();
			this->firstRun = true;
		}

		if (noSimCard)
		{
			DebugService::Warning("GSMConnection::loop", " No simcard!");
			this->connected = false;
			return;
		}

		if (this->connected == false || !this->modem.isGprsConnected())
		{

			DebugService::Information("GSMConnection::loop", " recycling connection!");

			String registrationStatus = String(this->modem.getRegistrationStatus());
			DebugService::Information("GSMConnection::loop Network Mode:", registrationStatus);

			DebugService::Information("GSMConnection::loop", " waiting for network!");
			if (!this->modem.waitForNetwork(10000L))
			{
				DebugService::Warning("GSMConnection::loop", "Could not get network!");
				this->connected = false;
				failCounter--;
			}
			else {
				DebugService::Information("GSMConnection::loop", "Connecting to " + apnAdress);
				DebugService::Information("GSMConnection::loop", "Operator " + String(this->modem.getOperator()));
				DebugService::Information("GSMConnection::loop", "Signal strength " + String(this->modem.getSignalQuality()));

				this->client.setTimeout(30000);
				if (this->modem.gprsConnect(apnAdress.c_str(), apnUser.c_str(), apnPassword.c_str()))
				{
					this->connected = true;
				}
			}
			if (failCounter == 0)
			{
				this->RestartModem();
				failCounter = 100;
			}

		}
		else
		{
			this->modem.maintain();
			failCounter = 100;
			this->connected = true;
		}
	}
}

Client* GSMConnection::getClient()
{
	return &this->client;
}

long GSMConnection::getEpochTime()
{
	return getNTPTime();
}

long GSMConnection::getNTPTime()
{
	modem.sendAT(GF("+CNTPCID=1"));
	modem.waitResponse();

	DebugService::Information("GSMConnection::getNTPTime","Sync time with: " + ConfigurationService::NTPUrl);

	modem.sendAT(GF("+CNTP=\""+ ConfigurationService::NTPUrl +"\",-12"));
	modem.waitResponse();

	modem.sendAT(GF("+CNTP"));
	modem.waitResponse(3000, GF("+CNTP: 1" GSM_NL));

	modem.sendAT(GF("+CCLK?"));
	modem.waitResponse(GF(GSM_NL));
	String cclk = modem.stream.readStringUntil('\n');
	modem.waitResponse();
	DebugService::Information("GSMConnection::getNTPTime", "\r\nCCLK: " + String(cclk));

	String currentDt = cclk.substring(cclk.indexOf("\"") + 1, cclk.lastIndexOf("\""));
	currentDt.trim();
	DebugService::Information("GSMConnection::getNTPTime", "\r\ncurrentDt: " + String(currentDt));

	int hh, mm, ss, yy, mon, day, tz;
	struct tm when = { 0 };
	time_t epoch = 0;

	sscanf(currentDt.c_str(), "%d/%d/%d,%d:%d:%d-%d", &yy, &mon, &day, &hh, &mm, &ss, &tz);
	when.tm_hour = hh;
	when.tm_min = mm;
	when.tm_sec = ss;
	when.tm_year = 2000 + yy - 1900;

	when.tm_mon = mon - 1;
	when.tm_mday = day;
	when.tm_isdst = -1;
	epoch = mktime(&when);

	epoch = epoch + (60 * (tz / 4) * 60);
	DebugService::Information("GSMConnection::getNTPTime", "CURRENT EPOCH: " + String(epoch));

	return epoch;
}`

doom-fr, 2023-01-12:
On SIM7000G, I found that mode 51 does not work well (mode 2 also).
```
2 Automatic
13 GSM only
38 LTE only
51 GSM and LTE only
```
So now, I use only GSM only (13) or LTE only (38).

In LTE mode, I also use `AT+CMNB=` command to choose catm or nbiot (1 for catm, 2 for nbiot)

Not tested on SIM7080G, hope it can help !

capedra, 2023-05-23:
I've been facing this registering issue with this module lately and whenever it stops registering on the network operators of my country, I simply downgrade/upgrade its firmware and suddenly it mysteriously registers again on these networks. The only clue that I have about this is that it may be related to some power issue or something.
But anyway, I hope it can help anyone facing this same issue before throwing these modules in the trash.

TheRealNapalm, 2023-06-15:
Hi, 
I have about 20 modems where AT+CGMR returns
Revision:1951B05SIM7080
and about 10 modems where the same command returns
Revision:1951B10SIM7080

The B10 firmware seems to be the newer one.
So all of  the B05 modems work fine with the german sim cards I am using. AT+COPS=1,2,"2... works fine. They all connect to a network within very short time. All is fine.

But, non of the B10 firmware modems can connect to any network using the same sim cards. Doing a AT+COPS=1,2,"2... command results in a 'No Service, On line' result.

I tried every possible solution from every post above, without any success.

I guess the only solution seems to update the firmware, but the modems are soldered and as far as i know i would have to connect a usb cable directly to the pins 24-27 of the SIM7080G chip, which is not possible for me in this constellation.

Is there a way to update or better downgrade the firmware over a normal serial connection?

fgnievinski, 2023-06-19:
here it's suggested to use the command CGDCONT to fix the PDP:
https://github.com/vshymanskyy/TinyGSM/issues/421#issuecomment-664029242

Olivier-zitoun, 2023-06-19:
Hi all,
For me, the command AT+CGDCONT=1,"IPV4V6",""  solved the problem.

kjm1102, 2024-01-17:
> Hi, I have about 20 modems where AT+CGMR returns Revision:1951B05SIM7080 and about 10 modems where the same command returns Revision:1951B10SIM7080
> 
> The B10 firmware seems to be the newer one. So all of the B05 modems work fine with the german sim cards I am using. AT+COPS=1,2,"2... works fine. They all connect to a network within very short time. All is fine.
> 
> But, non of the B10 firmware modems can connect to any network using the same sim cards. Doing a AT+COPS=1,2,"2... command results in a 'no network service' result and modem.waitForNetwork never finds a network.
> 
> I tried every possible solution from every post above, without any success.
> 
> I guess the only solution seems to update the firmware, but the modems are soldered and as far as i know i would have to connect a usb cable directly to the pins 24-27 of the SIM7080G chip, which is not possible for me in this constellation.
> 
> Is there a way to update or better downgrade the firmware over a normal serial connection?

I've had a similar experience here in AUS where some sims can do http only with 1951B07SIM7080 firmware. Sadly the native USB seems to be the only way to load new firmware. There is supposed to be a FOTA alternative where the difference between firmwares is used but in practice this service is unavailable. My problem is the only firmware readily available is the latest (1951B14Sim7080) so I can't roll back to 1951B07SIM7080. Wish I'd never upgraded B07 to B14, "if it ain't broke don't fix it"!

NavyaDevB, 2024-04-17:
Hi everyone, I am using TTGO SIM7000G ESP32 WROOVER-E Development board. I am facing a issue. I want to send some data to the cloud using MQTT.
In setup:
The code restarts the modem, connects to GSM, GPRS and MQTT.
In loop:
The code checks for GSM, GPRS and MQTT connectivity, collects the data and then publishes to the broker.
In the loop, at the start of the loop, I have given serial print("In loop").

Problem:
When I reset the device, it executes the setup and coming to setup - it only prints In loop for a number of times. Ex: from 5 to 10 times. 
Then [40968](ex:"[Some number]") Closed: 0.

Can anyone help me with it.

mwasela, 2024-05-24:
> and

It is indeed a power issue, I have a lily go with sim 7080, and am always stuck in the registration loop. I still cant get it to work. I was in touch with my ISP, and they could see the registration requests on their backend for my simcards IMSI, but the connection was rejected due to low power/network signal power. Am still trying out above pointed issues to get it to work.

maleeqB, 2024-06-10:
> On SIM7000G, I found that mode 51 does not work well (mode 2 also).
> 
> ```
> 2 Automatic
> 13 GSM only
> 38 LTE only
> 51 GSM and LTE only
> ```
> 
> So now, I use only GSM only (13) or LTE only (38).
> 
> In LTE mode, I also use `AT+CMNB=` command to choose catm or nbiot (1 for catm, 2 for nbiot)
> 
> Not tested on SIM7080G, hope it can help !
> 
> PS : I'm not a user of TinyGSM, just I pass here

fixed it for me (AT+CNMP) on SIM7000G. Thank you very much for your input

ctrewartha, 2025-08-22:
In case this helps, I had similar issues using SIM7080G, but found I had to set the APN first using CGDCONT command before the device was allowed to register on the network.