vshymanskyy/TinyGSM #598: https post sim7600 SSL

hmzaaa, 2021-10-15:
Hello,
please can someone help me.
I have tested this code works fine https://github.com/kiddtang/SSLClient/tree/master/examples/Esp32/https_gsm_SIM7600/https_post_sim7600
 
but when i try to change the server to my own server it doesn't work

```
// Server details to test TCP / SSL
const char server [] = "device-node-deploy.herokuapp.com";
const char resource [] = "/ device";
const int port = 443;
```

And the error i get:

```
Signal quality: 31
[1369] Current Network Time: 10/21 / 16.00: 26: 36
[1487] Modem Temp: 27.00
[1510] Connecting to memohamza.000webhostapp.com
Making POST request securely
[2162] ### Closed: 0
Status code: -2
Response:
[39366] ### Unhandled: + NETOPEN: 0

```
do we need to change the certificate

SRGDamia1, 2021-11-24:
Has this issue been resolved?

hmzaaa, 2021-11-24:
no not yet :/

simkard69, 2022-01-16:
Any update on the ability to use SSL/TLS communication with a remote server with SIM7600 ?
Seems like the Lilygo-T-SIM7600X has an example for that : [https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7600X/blob/master/examples/SSL_server/SSL_server.ino](url)

Any chance to get that running with TinyGSM ?

Naheel-Azawy, 2022-09-22:
Hello,

I haven't tried the fork @hmzaaa is using, but I tested the original one at https://github.com/govorox/SSLClient and it's working well so far.

> do we need to change the certificate

I guess you mean the one in `certs.h` in the provided example?
If yes, then probably this is your issue. 

Each domain must have a certificate. These certificates are signed by certificate authorities (CAs).
To check who is the root CA of a domain (I'll assume a chromium-based browser):
- Go to the lock next to the URL of the server you're trying to access
- "Connection is secure" -> "Certificate is valid"
- Under the "Details" tab, find the first in the hierarchy (e.g. "Default Trust:Magic Root 1")
- Go to something like `chrome://settings/certificates` and under "Authorities"
- Find "Magic" and then "Default Trust:Magic Root 1"
- Export to get the certificate you need and replace it with the one in your `certs.h`

For your server `device-node-deploy.herokuapp.com`, the CA seems to be `Amazon Root CA 1`.
So, I guess your `certs.h` should look like this

```c
const char* root_ca = \
    "-----BEGIN CERTIFICATE-----\n" \
    "MIIDQTCCAimgAwIBAgITBmyfz5m/jAo54vB4ikPmljZbyjANBgkqhkiG9w0BAQsF\n" \
    "ADA5MQswCQYDVQQGEwJVUzEPMA0GA1UEChMGQW1hem9uMRkwFwYDVQQDExBBbWF6\n" \
    "b24gUm9vdCBDQSAxMB4XDTE1MDUyNjAwMDAwMFoXDTM4MDExNzAwMDAwMFowOTEL\n" \
    "MAkGA1UEBhMCVVMxDzANBgNVBAoTBkFtYXpvbjEZMBcGA1UEAxMQQW1hem9uIFJv\n" \
    "b3QgQ0EgMTCCASIwDQYJKoZIhvcNAQEBBQADggEPADCCAQoCggEBALJ4gHHKeNXj\n" \
    "ca9HgFB0fW7Y14h29Jlo91ghYPl0hAEvrAIthtOgQ3pOsqTQNroBvo3bSMgHFzZM\n" \
    "9O6II8c+6zf1tRn4SWiw3te5djgdYZ6k/oI2peVKVuRF4fn9tBb6dNqcmzU5L/qw\n" \
    "IFAGbHrQgLKm+a/sRxmPUDgH3KKHOVj4utWp+UhnMJbulHheb4mjUcAwhmahRWa6\n" \
    "VOujw5H5SNz/0egwLX0tdHA114gk957EWW67c4cX8jJGKLhD+rcdqsq08p8kDi1L\n" \
    "93FcXmn/6pUCyziKrlA4b9v7LWIbxcceVOF34GfID5yHI9Y/QCB/IIDEgEw+OyQm\n" \
    "jgSubJrIqg0CAwEAAaNCMEAwDwYDVR0TAQH/BAUwAwEB/zAOBgNVHQ8BAf8EBAMC\n" \
    "AYYwHQYDVR0OBBYEFIQYzIU07LwMlJQuCFmcx7IQTgoIMA0GCSqGSIb3DQEBCwUA\n" \
    "A4IBAQCY8jdaQZChGsV2USggNiMOruYou6r4lK5IpDB/G/wkjUu0yKGX9rbxenDI\n" \
    "U5PMCCjjmCXPI6T53iHTfIUJrU6adTrCC2qJeHZERxhlbI1Bjjt/msv0tadQ1wUs\n" \
    "N+gDS63pYaACbvXy8MWy7Vu33PqUXHeeE6V/Uq2V8viTO96LXFvKWlJbYK8U90vv\n" \
    "o/ufQJVtMVT8QtPHRh8jrdkPSHCa2XV4cdFyQzR1bldZwgJcJmApzyMZFo6IQ6XU\n" \
    "5MsI+yMRQ+hDKXJioaldXgjUkK642M4UwtBV8ob2xJNDd2ZhwLnoQdeXeGADbkpy\n" \
    "rqXRfboQnoZsG4q5WTP468SQvvG5\n" \
    "-----END CERTIFICATE-----\n";
```

There are probably better ways to get those certificates. But this was a quick dirty way that got the job done for me.