vshymanskyy/TinyGSM #603: Quectel MC60 HTTPClient problem

ta2bss, 2021-10-30:
Hi,
I hope this is the proper category.

I have an Arduino Mega based development board that works with Atmel Atmega 2560 processor and contains internal Quectel MC60 GSM / GNSS module

The wire connections are:

MC60_Tx -- D15
MC60_Rx -- D14
MC_RI -- D2
MC60_DTR-- D4
MC60--POWERKEY -- A15

I'm using HttpClient.ino which comes with TinyGSM
The board can connect to network, update Network Time but can't complete the HTTP GET request.

I'm using SerialAT Serial3 setting.
I think Tx and Rx and Power Pins (digitalWrite (A15,HIGH) line at Setup ).  apn , user and pass  are correctly set. Can it be the problem with the other pins?
I Couldn't find anywhere where can I set these pins.

Thanks.

SRGDamia1, 2021-11-24:
Has this issue been resolved?

If you're getting any AT back and forth, it's probably not a serial port issue.

alcho75, 2025-02-15:
I have a similar issue. I found a bug in TinyGsmClientMC60.h.

// FOUND ISSUES
// file: TinyGsmClientMC60.h at line 430
// original: (TinyGsmIpFromString(host) == IPAddress(0, 0, 0, 0) ?       0 : 1));
// corrected: (TinyGsmIpFromString(host) == IPAddress(0, 0, 0, 0) ?      1 : 0));