vshymanskyy/TinyGSM #605: why there is no one response to our problem

MahmoudMohamedAli, 2021-11-02:
why there is no one response to our fucking problem. u made just fucking pull shit library and throw it to our fucking face why you are so fucking sheet

star297, 2021-11-03:
Bad language perhaps? My mum said she would wash my mouth out with soap and water if I spoke like that, well actually not as bad as your comments tbh.

You have to understand that contributors to this project earn no money from this and are trying to help EVERYONE by sharing their hard work.
  
Perhaps Mr. MohamedMahmoudAliPonjabiKebab would like to write his own GSM library then we could add some comments about his work.

MahmoudMohamedAli, 2021-11-03:
OKAY,CAN I THEN TALK TO YOUR MOTHER TO TAKE DATE from her please give me her number. I promise i will try to forget her the world dude

MahmoudMohamedAli, 2021-11-04:
i'm still waiting her number dude!i promise you again i will make her dive in milk

SRGDamia1, 2021-11-24:
@star297 Thank you.

closed 2021-11-24 by SRGDamia1