vshymanskyy/TinyGSM #608: SIM7000G : unexpected "### Unexpected module reset!"

chris4git, 2021-11-07:
Hi,
after calling modem.waitForNetwork(180000L) the library displays

```
[32743] ### Unexpected module reset!
[32743] ### TinyGSM Version: 0.11.4
[32743] ### TinyGSM Compiled Module:  TinyGsmClientSIM7000
```
and then continues.

Trying to understand,  I look at the source file TinyGsmClientSIM7000.h (line 487+) and can't see  why init() is called after receiving "SMS Ready" :

```
   } else if (data.endsWith(GF(GSM_NL "SMS Ready" GSM_NL))) {
          data = "";
          DBG("### Unexpected module reset!");
          init();
        }
```

Shouldn't init() be outside of these parenthesizes ?

An explanation ?

SRGDamia1, 2021-11-24:
Are you still having this issue?

The SIM7000 spits out that `SMS Ready` message when it boots up.  If you're seeing it in your debug logs, it's probably because your modem is resetting itself somewhere in your waiting period.  If the library sees the reboot, it re-runs the init.  If I remember correctly from my testing this was mostly needed because when the modem reset unexpectedly it often would boot back up with command echo turned on which would screw up later communication in the library.

chris4git, 2021-11-24:
hi,

not really an issue as waitForNetwork returns ok and the execution continues...

SRGDamia1, 2021-11-24:
Ok, if it's not an issue, please close this.

closed 2021-11-26 by chris4git

tronar, 2025-01-02:
Well, no, but just for the record, I have a SIM7000 that even after sending SMS READY, it remains "unregistered":
at+cfun=0
+CPIN: NOT READY

OK
at+cfun=1
OK

+CPIN: READY

SMS Ready
at+creg?
+CREG: 0,2

OK

Larzu, 2025-10-25:
I do have same problem with SIM7080 which is sending SMS ready when  getting up.

Is there any solutions to get around this problem?