vshymanskyy/TinyGSM #609: Can't make calls with SIM7000E

axwell, 2021-11-09:
[ ] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[*] Question or request for help

### What are you working with?

Modem: SIM7000E
Main processor board: ESP32
TinyGSM version: latest
Code: 

### Scenario, steps to reproduce

I'm trying to make calls or send USSD but both fails with ERROR.
SMS is working sending / receving, internet & others.
Module is registered to the network.
If i'm trying to call the SIM7000e, robot says that is not online at the moment.

### Expected result

Call to work

### Actual result
ERROR:

### Debug and AT command log

SRGDamia1, 2021-11-24:
Are you still having issues?

ishdemon, 2022-01-12:
@axwell You got SMS working on 7000?? i have SIM7000G , USSD/ SMS doesn't work for me. Did you used code from example or somewhere else?

hossamxstudios, 2024-05-15:
@axwell i want to ask you if you were able to solve this problem

axwell, 2024-05-15:
@hossamxstudios i did not, gave up with this type of modem, i suspect it does not have this feature buitin.