vshymanskyy/TinyGSM #611: SIM7000G lilygo to AWS IOT CORE MQTT

TECA-IOT, 2021-11-09:
Hello,
Please I need to connect a sim7000G to AWS IoT Core, I am using the Tinygsm library. I can do simple MQTT communication, but I can't find any documentation or tutorial to load the certificates "CA", "XXXXXXXX-certificate.pem.crt" and "XXXXXXXX-private.pem.key"

Many in forums that look for the same thing but there is no clear documentation. I will appreciate it. Thanks!

SRGDamia1, 2021-11-24:
I'm sorry, I don't have any experience yet with uploading AWS certificates.

OussamaNRCI, 2022-02-12:
Hi @TECA-IOT 

Did you find how to load the certificate and connect the module with AWS IoT Core !!

RosiersRobin, 2022-03-03:
In need of this too!

zibyan123, 2022-09-22:
I am also trying to connect sim7000g with aws but I am not able to convert certificates .Is there any solution?

andygrillo, 2022-10-16:
Any updates on this ?

dankpool, 2023-09-12:
did anyone get how to connect with tinygsm library to aws iot core

yuzukihironaka-lefixea, 2023-10-02:
Hello,

We could connect to AWS IoT Core by wrapping TinyGsm with some SSLClient Library.

In my project, I use https://github.com/alkonosst/SSLClientESP32 library.

- Example

```cpp
#include <TinyGsm.h>
#include <SSLClientESP32.h>

TinyGsm modem(SerialAT);
TinyGsmClient base_client(modem, 0);
// wrap TinyGsmClient with SSLClient
SSLClientESP32 ssl_client(&base_client);
PubSubClient mqtt(ssl_client);

void setup() {
    ...
    // set certificates
    sslclient.setCACert(rootca);
    sslclient.setCertificate(certificate);
    sslclient.setPrivateKey(privatekey);
    
    // connect to broker
    mqtt.setServer(broker, port);
    mqtt.connect();
}
```

Basicly, It works fine.

But I also have issue https://github.com/alkonosst/SSLClientESP32/issues/9

I would like to discuss this issue for make stuff more stable :(

NavyaDevB, 2024-12-12:
Hi. Can someone help me in connecting SIM7000G to AWS IoT core. 
I have done till some steps. i.e., configuring ssl certificates but whenever I am trying to check the coongunapaj