vshymanskyy/TinyGSM #612: help to get TinyGSM to work with raw ESP-MQTT

garudaonekh, 2021-11-16:
Hi,
How to get this library work with raw ESP-MQTT?

My existing project is using [](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/mqtt.html) with Arduino IDE. But I can't get TinyGSM to work with it.

Thanks;

SRGDamia1, 2021-11-24:
I'm sorry if I'm misunderstanding what you're asking, but I'm pretty sure what you're asking isn't possible.

This library works via AT commands.  

The ESP32 (or ESP8266) can be used with TinyGSM:
- either running TinyGSM to talk to a *secondary cellular modem* (like a SIM800 or SIM7000) by way of serial AT commands
- or as a WiFi modem for another primary processor.  In this second case, the *other processor* is running TinyGSM and the Espressif chip must be running the "ESP-AT" firmware documented here:  https://docs.espressif.com/projects/esp-at/en/latest/index.html.

The ESP-MQTT component of the ESP-IDF framework is meant for the Espressif chip using MQTT on it's own built-in WiFi radio while the Espressif chip is being used as the primary processor.  So it's pretty much a completely different use case.  It's a really common source of confusion.

garudaonekh, 2021-11-25:
Thanks
I use PPPOS+ESP-MQTT instead of tinyGSM

closed 2021-11-26 by garudaonekh