vshymanskyy/TinyGSM #616: Check modem.isNetworkConnected() severally make SIM800L disconnect to network or GPRS

arimukhlas, 2021-11-26:
**[ ] Question or request help**

Hi skyy dan team, thanks for creating a very good library

I'm trying to send data to IOT platform [demo.thingsboard.io](https://demo.thingsboard.io) using SIM800L v2
I added procedure to [sample sketch](https://github.com/thingsboard/thingsboard-arduino-sdk/tree/master/examples/0004-arduino-sim900_send_telemetry) from thingsboard library

the procedure use 4 function before send data inside `loop()`
`modem.waitForNetwork()`
`modem.isNetworkConnected()`
`modem.isGprsConnected()`
`modem.gprsConnect()`

if all function fulfilled then send data
but if modem error or time out, i restart the modem sim800L

the problem is, after one success loop, when I check `modem.isNetworkConnected()`, the module SIM800L responded DISCONNECT until modem restart, is there any way to stable my procedure

this a small part of my program :

```
void loop() {
delay(1000);
  if(!modem.isNetworkConnected()){
      Serial.println("NETWORK DISCONNECTED!!!");

      //ReconnectingNetwork///////////////////////////////////////
      delay(1000);
      Serial.println("");
      Serial.println("MODEM IS RECONNECTING TO NETWORK...");
      if(!modem.waitForNetwork()){
        Serial.println("RECONNECTING TO NETWORK... FAIL...["+(String)attemptRestartModem1+"]");
        attemptRestartModem1++;
        
        if(attemptRestartModem1 > MAX_ATTEMPT){
          Serial.println("");
          Serial.println("RESTARTING MODEM...");
          attemptRestartModem1 = 0;
          delay(2000);
          modem.poweroff();
        }
        return;
      } 
      else {
        Serial.println("CONNECTED TO NETWORK : " + modem.getOperator());
        attemptRestartModem1 = 0;
        delay(1000);
      }
  }
  else {
    Serial.println("STILL CONNECTED TO NETWORK : " + modem.getOperator());
    delay(1000);
  }

  delay(1000);
  
  if (!modem.isGprsConnected()) {

    //ReconnectingGPRS//////////////////////////////////////////
    Serial.println("GPRS DISCONNECTED!!!");
    delay(1000);
    Serial.println("");
    Serial.println("MODEM IS RECONNECTING TO NETWORK\'S GPRS...");
    
    if (!modem.gprsConnect(apn.c_str())) {
      Serial.println("RECONNECTING NETWORK\'S GPRS... FAIL...");
      Serial.println("");
      Serial.print("MODEM RESTART IN ");
      delay(1000);
      Serial.print("3 . . ");
      delay(1000);
      Serial.print("2 . . ");
      delay(1000);
      Serial.println("1");
      delay(2000);
      modem.poweroff();
      delay(2000);
      ESP.restart();
    }
    else {
      Serial.println("CONNECTED TO NETWORK\'S GPRS : " + String(apn));
      Serial.println("NEW IP ADDRESS : " + modem.getLocalIP());
      Serial.println("");
      delay(1000);
    }
  }
  else {
    Serial.println("STILL CONNECTED TO GPRS!!! " + modem.getLocalIP());
    Serial.println("");
  }

//Connect to the ThingsBoard///////////////////////////////////////////
  if (!tb.connected()) {
    Serial.print("Connecting to: ");
    Serial.print(yourinputString1);
    Serial.print(" with token ");
    Serial.println(yourinputString2);
    if (!tb.connect(yourinputString1.c_str(), yourinputString2.c_str())) {
      
      Serial.println("Failed to connect..."+(String)attemptRestartESP);
      attemptRestartESP++;
      if(attemptRestartESP >= MAX_ATTEMPT){
        Serial.println("Restart ESP...");
        delay(2000);
        ESP.restart();
      }
      return;
    }
  }
  
  attemptRestartESP = 0;

  delay(2000);

  long data1 = 100;
  
  tb.sendTelemetryFloat("pressure-003", (float)data1/100);

  delay(5000);
  data1=data1+10;

}
```

_note : 
I use bidirrectional 3.3V to 5V TTL converter module to communicate Wemos D1 Mini Pro and SIM800L v2
AT-Command is stable using 9600 Baudrate
I use 2A charger to power the modem module_

SRGDamia1, 2021-11-29:
Hm. Can you post an AT log?  Quickly skimming your code, it seems like it *should* work...

arimukhlas, 2021-12-01:
> Hm. Can you post an AT log? Quickly skimming your code, it seems like it _should_ work...

how to do that (AT log)??