vshymanskyy/TinyGSM #634: gprsConnect suddenly stack sometimes in first initial connection

roysG, 2022-01-19:
I am using in the latest version of TinyGsm.
I am using sim7070g in esp32

As you can see in the logs in the bottom, the code is stack over and over again...

This are the errors i get:

```
E (87757) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (87757) task_wdt:  - IDLE (CPU 1)
E (87757) task_wdt: Tasks currently running:
E (87757) task_wdt: CPU 0: IDLE
E (87757) task_wdt: CPU 1: loopTask
E (87757) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x400ED3BB:0x3FFB0B900x40082C51:0x3FFB0BB0 0x40155AE3:0x3FFC6CA0 0x400D366A:0x3FFC6CC0 0x4008A7B5:0x3FFC6CE0 0x4008C4D5:0x3FFC6D00
0x400ed3bb: task_wdt_isr at /Users/roysabah/esp/esp-idf/components/esp_system/task_wdt.c:182 (discriminator 3)

0x40082c51: _xt_lowint1 at /Users/roysabah/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1111

0x40155ae3: cpu_ll_waiti at /Users/roysabah/esp/esp-idf/components/hal/esp32/include/hal/cpu_ll.h:183
 (inlined by) esp_pm_impl_waiti at /Users/roysabah/esp/esp-idf/components/esp_pm/pm_impl.c:830

0x400d366a: esp_vApplicationIdleHook at /Users/roysabah/esp/esp-idf/components/esp_system/freertos_hooks.c:63

0x4008a7b5: prvIdleTask at /Users/roysabah/esp/esp-idf/components/freertos/tasks.c:3954 (discriminator 1)

0x4008c4d5: vPortTaskWrapper at /Users/roysabah/esp/esp-idf/components/freertos/port/xtensa/port.c:130

E (87757) task_wdt: Print CPU 1 backtrace

Backtrace:0x40084D7D:0x3FFB11900x40082C51:0x3FFB11B0 0x4000BFED:0x3FFC92A0 0x4008C72E:0x3FFC92B0 0x4008A180:0x3FFC92D0 0x400DFB9D:0x3FFC9310 0x400E0229:0x3FFC9340 0x400D8154:0x3FFC9360 0x400D9ABE:0x3FFC9480 0x400DA573:0x3FFC94C0 0x400E058A:0x3FFC9610 0x4008C4D5:0x3FFC9630
0x40084d7d: esp_crosscore_isr at /Users/roysabah/esp/esp-idf/components/esp_system/crosscore_int.c:92

0x40082c51: _xt_lowint1 at /Users/roysabah/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1111

0x4008c72e: vPortClearInterruptMaskFromISR at /Users/roysabah/esp/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:546
 (inlined by) vPortExitCritical at /Users/roysabah/esp/esp-idf/components/freertos/port/xtensa/port.c:307

0x4008a180: xQueueSemaphoreTake at /Users/roysabah/esp/esp-idf/components/freertos/queue.c:1612

0x400dfb9d: uartAvailable at /Users/roysabah/esp/esp32-arduino-lib-builder/components/arduino/cores/esp32/esp32-hal-uart.c:192 (discriminator 1)

0x400e0229: HardwareSerial::available() at /Users/roysabah/esp/esp32-arduino-lib-builder/components/arduino/cores/esp32/HardwareSerial.cpp:193

0x400d8154: TinyGsmSim7080::waitResponse(unsigned int, String&, char const*, char const*, char const*, char const*, char const*) at /Users/roysabah/esp/esp32-arduino-lib-builder/components/arduino/libraries/TinyGSM/src/TinyGsmClientSIM7080.h:612 (discriminator 1)

0x400d9abe: TinyGsmSim7080::waitResponse(unsigned int, char const*, char const*, char const*, char const*, char const*) at /Users/roysabah/esp/esp32-arduino-lib-builder/components/arduino/libraries/TinyGSM/src/TinyGsmClientSIM7080.h:709
 (inlined by) TinyGsmSim7080::gprsConnectImpl(char const*, char const*, char const*) at /Users/roysabah/esp/esp32-arduino-lib-builder/components/arduino/libraries/TinyGSM/src/TinyGsmClientSIM7080.h:273

0x400da573: TinyGsmSim70xx<TinyGsmSim7080>::gprsConnectImpl(char const*, char const*, char const*) at /Users/roysabah/esp/esp32-arduino-lib-builder/components/arduino/libraries/TinyGSM/src/TinyGsmClientSIM70xx.h:249
 (inlined by) TinyGsmGPRS<TinyGsmSim70xx<TinyGsmSim7080> >::gprsConnect(char const*, char const*, char const*) at /Users/roysabah/esp/esp32-arduino-lib-builder/components/arduino/libraries/TinyGSM/src/TinyGsmGPRS.tpp:54
 (inlined by) setup() at /Users/roysabah/esp/esp32-arduino-lib-builder/main/sketch.cpp:1704

0x400e058a: loopTask(void*) at /Users/roysabah/esp/esp32-arduino-lib-builder/components/arduino/cores/esp32/main.cpp:38

0x4008c4d5: vPortTaskWrapper at /Users/roysabah/esp/esp-idf/components/freertos/port/xtensa/port.c:130

E (92757) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (92757) task_wdt:  - IDLE (CPU 1)
E (92757) task_wdt: Tasks currently running:
E (92757) task_wdt: CPU 0: IDLE
E (92757) task_wdt: CPU 1: loopTask
E (92757) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x400ED3BB:0x3FFB0B900x40082C51:0x3FFB0BB0 0x40155AE3:0x3FFC6CA0 0x400D366A:0x3FFC6CC0 0x4008A7B5:0x3FFC6CE0 0x4008C4D5:0x3FFC6D00
0x400ed3bb: task_wdt_isr at /Users/roysabah/esp/esp-idf/components/esp_system/task_wdt.c:182 (discriminator 3)

0x40082c51: _xt_lowint1 at /Users/roysabah/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1111

[... 666 lines cut ...]

Backtrace:0x40084D7D:0x3FFB11900x40082C51:0x3FFB11B0 0x40089B3B:0x3FFC92D0 0x400DFBC1:0x3FFC9310 0x400E0229:0x3FFC9340 0x400D8154:0x3FFC9360 0x400D9ABE:0x3FFC9480 0x400DA573:0x3FFC94C0 0x400E058A:0x3FFC9610 0x4008C4D5:0x3FFC9630
0x40084d7d: esp_crosscore_isr at /Users/roysabah/esp/esp-idf/components/esp_system/crosscore_int.c:92

0x40082c51: _xt_lowint1 at /Users/roysabah/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1111

0x40089b3b: xQueueGenericSend at /Users/roysabah/esp/esp-idf/components/freertos/queue.c:825 (discriminator 2)

0x400dfbc1: uartAvailable at /Users/roysabah/esp/esp32-arduino-lib-builder/components/arduino/cores/esp32/esp32-hal-uart.c:196

0x400e0229: HardwareSerial::available() at /Users/roysabah/esp/esp32-arduino-lib-builder/components/arduino/cores/esp32/HardwareSerial.cpp:193

0x400d8154: TinyGsmSim7080::waitResponse(unsigned int, String&, char const*, char const*, char const*, char const*, char const*) at /Users/roysabah/esp/esp32-arduino-lib-builder/components/arduino/libraries/TinyGSM/src/TinyGsmClientSIM7080.h:612 (discriminator 1)

0x400d9abe: TinyGsmSim7080::waitResponse(unsigned int, char const*, char const*, char const*, char const*, char const*) at /Users/roysabah/esp/esp32-arduino-lib-builder/components/arduino/libraries/TinyGSM/src/TinyGsmClientSIM7080.h:709
 (inlined by) TinyGsmSim7080::gprsConnectImpl(char const*, char const*, char const*) at /Users/roysabah/esp/esp32-arduino-lib-builder/components/arduino/libraries/TinyGSM/src/TinyGsmClientSIM7080.h:273

0x400da573: TinyGsmSim70xx<TinyGsmSim7080>::gprsConnectImpl(char const*, char const*, char const*) at /Users/roysabah/esp/esp32-arduino-lib-builder/components/arduino/libraries/TinyGSM/src/TinyGsmClientSIM70xx.h:249
 (inlined by) TinyGsmGPRS<TinyGsmSim70xx<TinyGsmSim7080> >::gprsConnect(char const*, char const*, char const*) at /Users/roysabah/esp/esp32-arduino-lib-builder/components/arduino/libraries/TinyGSM/src/TinyGsmGPRS.tpp:54
 (inlined by) setup() at /Users/roysabah/esp/esp32-arduino-lib-builder/main/sketch.cpp:1704

0x400e058a: loopTask(void*) at /Users/roysabah/esp/esp32-arduino-lib-builder/components/arduino/cores/esp32/main.cpp:38

0x4008c4d5: vPortTaskWrapper at /Users/roysabah/esp/esp-idf/components/freertos/port/xtensa/port.c:130
```

viniciusdesa, 2022-05-02:
I'm also getting same trouble:

`(319187) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (319187) task_wdt:  - IDLE (CPU 0)
E (319187) task_wdt: Tasks currently running:
E (319187) task_wdt: CPU 0: Gsm Connection 
E (319187) task_wdt: CPU 1: loopTask
E (319187) task_wdt: Aborting.
abort() was called at PC 0x40127834 on core 0

Backtrace:0x40084505:0x3ffbea000x400950f1:0x3ffbea20 0x4009a3b1:0x3ffbea40 0x40127834:0x3ffbeac0 0x40085695:0x3ffbeae0 0x40084445:0x3ffe7030 0x400966df:0x3ffe7050 0x4011b521:0x3ffe7070 0x400fa773:0x3ffe7090 0x400faa72:0x3ffe70f0 0x400fd0e2:0x3ffe7150 0x400fd29c:0x3ffe7190 0x400fc0d5:0x3ffe71b0 
ELF file SHA256: 0000000000000000
Rebooting...`

SRGDamia1, 2022-05-02:
I'm not familiar with the ESP32's watchdog system, but the GSM connection can take a long time (1-5+ minutes, depending on the modem).  This library blocks for the entire time.  If your watchdog fires too quickly, you'll get a reset.

The first connection with a new modem or a new tower often takes longer than subsequent connections.

erdebee, 2023-11-01:
You can set a yield time like so:

#define TINY_GSM_YIELD_MS 1

This will make the Task WDT reset itself.