vshymanskyy/TinyGSM #636: OTA Problem (help)

lucasromeiro, 2022-01-27:
Hi guys, I have a problem.

I use esp32 and a sim800l modem.
I need to do OTA update safely, so there is no risk of a device breaking with a failed update.
However, apparently the tinygsm communication object is not supported for the standard OTA update function.
I have a large program, almost 1mb of code.
I looked at people downloading the BIN file for SPIFFS and then updating. Is this 100% safe? Is there a risk of failure and killing equipment that is remote?
I tried to implement it but I couldn't because the . BIN is too big.

Maybe if I change the file system to 1M of program and 3M of SPIFFS. It is possible?

If there is another way, can you help me?

zark-zeugan, 2022-02-13:
Can I ask you if its is a HTTP or HTTPS request you are doing?

salvatorescarantino, 2022-02-15:
I have the same problem: with the 800KB file bin I haven't any problem, but with 1MB file BIN the download fails.
What's the problem?

lucasromeiro, 2022-02-15:
> Can I ask you if its is a HTTP or HTTPS request you are doing?

http

salvatorescarantino, 2022-02-15:
> > Can I ask you if its is a HTTP or HTTPS request you are doing?
> 
> http

me too

salvatorescarantino, 2022-02-15:
I've solved the problem: the main problem isn't the http request, but the writing of spiffs. In the example sketch there's a single while command that writing that client read. In this case at the end of download the command not work correctly and the while command stops. I've made 3 fix:
1. File file = SPIFFS.open("/update.bin", FILE_WRITE); --->I've change FILE_APPEND in FILE_WRITE
2. I've add more while command during download bin file;
3. Added a delay of 500ms in every serial transmission of percent.

Try it

zark-zeugan, 2022-02-16:
> I've solved the problem: the main problem isn't the http request, but the writing of spiffs. In the example sketch there's a single while command that writing that client read. In this case at the [...]

Could you share a snippet of the code? 
Also, what server are you using to store your binaries?

gotnull, 2023-01-14:
Any update on this?