vshymanskyy/TinyGSM #639: TinyGSM making "AT+SAPBR=2,1" call upon server get request seems to fail internally

Techmokid, 2022-02-02:
[x ] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
[ ] Bug or problem compiling the library
[ x] Bug or issue with library functionality (ie, sending data over TCP/IP)
[ x] Question or request for help

### What are you working with?

Modem: SIM7000G
Main processor board: Arduino Mega 2560
TinyGSM version: 0.11.5 (Latest as of this post)
Code: FAR too long to be put here

Hello!

I have been trying to get TinyGSM to work for awhile now, and most of it has juust been debugging my own code. However, I may have come across a bug in the library while diagnosing my code. When I start the SIM7000 and attempt to get data from a server of mine, I get the following output:
[Public-safe Arduino Output.txt](https://github.com/vshymanskyy/TinyGSM/files/7987659/Public-safe.Arduino.Output.txt)

Now, it seems to fail when calling "AT+SAPBR=2,1". For some reason, the returned data is:
+SAPBR: 1,1,"10.132.42.219

Notice the missing quotation mark. Resulting in the inevitable message:
[96460] ### Unhandled: +SAPBR: 1,1,"10.132.42.219

While my own code is smart enough to detect this failure and try again, this issue always comes up. I do not know why. Something about that dropped quotation mark at the end absolutely CRIPPLES this library in my case. Not sure if this is a known bug or what. I could post my code here if requested, but it is LONG and might end up confusing people more than it would help.

While my code not working is mostly down to just me making mistakes and bugs, this particular issue seems to be fault of the librbary instead of myself this time. I'm just curious if there is a known fix for this, or if it's a known bug currently being tracked, or what?

Hope to hear from someone about this soon,

Kind regards,
Andrey

fgnievinski, 2023-06-27:
I also notice that when issuing the command `AT+SAPBR=2,1`, the expected response is not just "OK", but more importantly, `+SAPBR: <cid>,<Status>,<IP_Addr>`:
https://github.com/vshymanskyy/TinyGSM/blob/5b83997877c96fe3388f634121d91ed02503728e/src/TinyGsmClientSIM7000.h#L204-L206

OmegaLeWolf, 2024-06-17:
Sorry about the bump, any solutions on this?