vshymanskyy/TinyGSM #641: SIM7600 suggest fix code for GPS visible satellites

chepil, 2022-02-08:
SIM7600E-H, TinyGSM v.0.11.5

Right now it always Zero result for Visible Satelites counter. 

AT Commands:
AT+CGNSSINFO
+CGNSSINFO: 2,05,00,00,6041.475467,N,02846.185792,E,080222,114731.0,85.9,0.0,,1.5,1.3,0.9

Then, when the parse getGPXImpl, "vsat" is ignored.
Code: from TinyGsmClientSIM7600.h.  (getGPSImpl)

is:
....
float secondWithSS = 0;

streamSkipUntil(',');               // GPS satellite valid numbers
streamSkipUntil(',');               // GLONASS satellite valid numbers
streamSkipUntil(',');               // BEIDOU satellite valid numbers
....
as far as I understand, correct code will be:
...
float secondWithSS = 0;

ivsat = streamGetIntBefore(',');
streamSkipUntil(',');               // GLONASS satellite valid numbers
streamSkipUntil(',');               // BEIDOU satellite valid numbers
...

Tested Devices is:
aliexpress.ru/item/1005001705250713.html

AirTobe91, 2022-03-09:
I would also like to get a update with a fix version. I have the same problem with showing the available and visible satellites.

sensuslankoylu, 2022-05-03:
i have same problem. because we have one more parameter in gps data and tinygsm lib cant parse it.

i changed like this.

`streamSkipUntil(',');               // GPS satellite valid numbers
      streamSkipUntil(',');               // GLONASS satellite valid numbers
      streamSkipUntil(',');               // BEIDOU satellite valid numbers
      streamSkipUntil(',');               // one not known parameter
      ilat  = streamGetFloatBefore(',');  // Latitude in ddmm.mmmmmm
      north = stream.readStringUntil(',').charAt(0);              // N/S Indicator, N=north or S=south
      ilon = streamGetFloatBefore(',');  // Longitude in ddmm.mmmmmm
      east = stream.readStringUntil(',').charAt(0);              // E/W Indicator, E=east or W=west
      // Date. Output format is ddmmyy
      iday   = streamGetIntLength(2);    // Two digit day
      imonth = streamGetIntLength(2);    // Two digit month
      iyear  = streamGetIntBefore(',');  // Two digit year

      // UTC Time. Output format is hhmmss.s
      ihour = streamGetIntLength(2);  // Two digit hour
      imin  = streamGetIntLength(2);  // Two digit minute
      secondWithSS =
          streamGetFloatBefore(',');  // 4 digit second with subseconds

      ialt   = streamGetFloatBefore(',');  // MSL Altitude. Unit is meters
      ispeed = streamGetFloatBefore(',');  // Speed Over Ground. Unit is knots.
      streamSkipUntil(',');                // Course Over Ground. Degrees.
      streamSkipUntil(',');  // After set, will report GPS every x seconds
      iaccuracy = streamGetFloatBefore(',');  // Position Dilution Of Precision
      streamSkipUntil(',');   // Horizontal Dilution Of Precision
      streamSkipUntil(',');   // Vertical Dilution Of Precision`

jabejaranoq, 2023-07-11:
I started working a TTGO AX7670 board with a A7670E modem and got wrong gps data and after digging a bit  found out the exact same problem as @sensuslankoylu  (ghost parameter in the result string).  I did also found out that this math is wrong:

        *lat = (floor(ilat / 100) + fmod(ilat, 100.) / 60) *
              (north == 'N' ? 1 : -1);

I made a comparision with the 7000G code (which is working in my 7000G board) and found out that the functional one is:

        *lat = ilat * (north == 'N' ? 1 : -1);

I applied same fix to the *lon pointer.

Looking into the SIMCom AT commands documentation for the 7600 series and the 76XX series discovered that the missing parameter is not reported there too (see image).  Looks like documentation needs update.

[image]

Regarding the original issue reported by @chepil I agree too with the fix.  

Looks like now my A7670E is getting valid GPS data applying fixes above.  I would like to submit a pull request for those changes but I am not sure if the appropiate is to modify TinyGsmClientSIM7600.h or create a new .h file for my specific A7670E modem.   I have not hardware to test other 7600 series modem.  Advice is welcome.