vshymanskyy/TinyGSM #644: A9G GSM/GPRS module library for esp32

ayazurrehman786, 2022-02-23:
How to use TinyGSM library for A9G module. Is it compatible with A9G to be used in ESP32 as http client.

praxnax, 2023-04-05:
Hi, are you still working on the project?

a-3isa, 2024-03-21:
Hi , can you help me i am trying to interface A9 with ESP32c3 ?

Mukund2900, 2024-06-13:
@a-3isa @praxnax @ayazurrehman786 did this work somehow ? Getting the netwrok establised ? 

I tried this 
```
#define TINY_GSM_MODEM_A7
#include <TinyGsmClient.h>

// Set your APN
const char apn[]  = "airtelgprs.com";  // type your APN
const char user[] = "";                // Leave empty if not required
const char pass[] = "";                // Leave empty if not required

// Server details
const char server[] = "dummy.restapiexample.com";
const int port = 80; // HTTP port

TinyGsm modem(Serial2);                // Use Serial2 for communication with A9G
TinyGsmClient client(modem);

void setup() {
  Serial.begin(115200);
  delay(500);
  Serial.println("\nInitializing modem...");
  Serial2.begin(115200);               // Initialize Serial2
  delay(2000);

  // Initialize modem using TinyGsm
  Serial.println("Initializing modem using TinyGsm...");
  if (!modem.restart()) {
    Serial.println("Failed to restart modem, attempting to initialize...");
    if (!modem.init()) {
      Serial.println("Failed to initialize modem, please check your connections and reset the board.");
      while (true);
    }
  }

  Serial.println("Modem initialized");

  Serial.println("Connecting to network...");
  if (!modem.waitForNetwork()) {
    Serial.println("Network connection failed");
    while (true);
  }

  Serial.println("Network connected");

  Serial.println("Connecting to GPRS...");
  if (!modem.gprsConnect(apn, user, pass)) {
    Serial.println("GPRS connection failed");
    while (true);
  }

  Serial.println("GPRS connected");
  makeHttpRequest();
}

void loop() {
  // Your code to use the internet connection goes here
}

void makeHttpRequest() {
  Serial.println("Connecting to server...");
  if (!client.connect(server, port)) {
    Serial.println("Connection to server failed");
    return;
  }

  Serial.println("Connected to server, sending GET request...");

  // Make an HTTP GET request
  client.println("GET /api/v1/employees HTTP/1.1");
  client.println("Host: dummy.restapiexample.com");
  client.println("Connection: close");
  client.println();

  // Read all the lines of the reply from server and print them to Serial
  while (client.connected() || client.available()) {
    if (client.available()) {
      String line = client.readStringUntil('\n');
      Serial.println(line);
    }
  }

  // Close the connection
  client.stop();
  Serial.println("HTTP request sent, connection closed");
}
```

But all it gives is 
Modem initialized
Connecting to network...
Network connection failed

Restart did happen but network is not getting established