vshymanskyy/TinyGSM #654: Sim 800c Stuck in +CREG: 0,5

Paulf007, 2022-04-13:
[X ] I have read the Troubleshooting section of the ReadMe

**What type of issues is this?**
[ ] Bug or issue with library functionality - not recognizing the response code 

**What are you working with?**
Modem: SIM800C
Main processor board: UNO Softserial
TinyGSM version: 0.11.5
Code: basic mqtt example 

**Scenario, steps to reproduce**
Basic startup , it detects the modem and then goes on the send the AT+CREG? command. The response confirms that the module is connected to the network -  +CREG: 0,5 but it stays in that loop until it returns a fail. 

**Expected result**
Having a Active MQTT Connection 

**Actual result**
Code is stuck in a loop when the "AT+CREG?" is send 

Debug and AT command log
[112034] ### Unhandled: 
AT+CREG?
AT+CREG?

+CREG: 0,5

OK
[113046] ### Unhandled: 
AT+CSQ
AT+CSQ

+CSQ: 25,5

OK
[114305] ### Unhandled: 
AT+CREG?

Wolfleader101, 2022-07-15:
Hey, did you end up resolving this issue?

ghazanhaider, 2023-01-17:
I've had similar issues with my SIM808 with TinyGSM verison 0.11.5

It hangs on checking CREG=0,1

Doing the CIP commands by hand works even with CREG=0,1

The trick was to change the test command to CGREG, in TinyGsmClientSIM800.h line 294:

public:
  RegStatus getRegistrationStatus() {
    return (RegStatus)getRegistrationStatusXREG("CREG");
  }

To

public:
  RegStatus getRegistrationStatus() {
    return (RegStatus)getRegistrationStatusXREG("CGREG");
  }