vshymanskyy/TinyGSM #679: PROBLEM #error "Please define GSM modem model" is already defined

Jhonattan38G, 2022-08-25:
hello everyone, I have an error and I still don't understand its origin, when I install the tinygsm library, everything goes well, but when I declare the TinyGsm and TinyGsmClientSecure objects it doesn't recognize them, I'll add screenshots to show, because I already declared the GSM model and still So it still doesn't work, as I said, I don't understand the origin of this error, I need help.
[image]
[image]
[image]
[image]

nkgiovannivl, 2022-09-05:
Hi,

In my case #define before #include works, i'm not sure if it was my problem.

[image]

chinswain, 2024-08-24:
Did you manage to resolve this?