vshymanskyy/TinyGSM #684: The ability to work with SMS.

psxbox, 2022-10-15:
[ ] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[ ] Bug or issue with library functionality (ie, sending data over TCP/IP)
[x] Question or request for help

### What are you working with?

Modem: SIM7600, SIM800
Main processor board: ESP32
TinyGSM version: 0.11.5
Code: 

Hello. Are you planning to add SMS-working (reading, deleting, etc.) methods to the library?

dharmalesmana16, 2022-10-20:
bro i need an answer , does TinyGSM library support for A7600C1 Module ? or simple question does tinygsm library support other modules besides those mentioned ?

psxbox, 2022-10-20:
Must support. Write #define TINY_GSM_MODEM_SIM7600 above. SIM7600 C E G they differ only in regions. AT commands are the same.

psxbox, 2022-10-20:
> bro i need an answer , does TinyGSM library support for A7600C1 Module ? or simple question does tinygsm library support other modules besides those mentioned ?

Must support. Write #define TINY_GSM_MODEM_SIM7600 above. SIM7600 C E G they differ only in regions. AT commands are the same.