vshymanskyy/TinyGSM #689: Modified SIM800 to work with SIM7020.

1mek, 2022-11-01:
[ ] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[ ] Bug or issue with library functionality (ie, sending data over TCP/IP)
[x] Question or request for help

### What are you working with?

Modem: SIM7020G
Main processor board: ESP32
TinyGSM version: latest (0.11.5)
Code: 
[TinyGsmClientSIM7020m.zip](https://github.com/vshymanskyy/TinyGSM/files/9908155/TinyGsmClientSIM7020m.zip)

In arduino-ide include above as local file in sketchfolder then in sketch. ```
#define TINY_GSM_MODEM_SIM7020
#include "TinyGsmClientSIM7020m.h"

typedef TinyGsmSim7020                   TinyGsm;
typedef TinyGsmSim7020::GsmClientSim7020 TinyGsmClient;
```

### Scenario, steps to reproduce
Ive been modifying Sim800 code to work with the sim7020, so far only tested with mqtt examples and it works fine.
But is there any downside using sim7020 with the sim800-compability commands?

capedra, 2023-01-25:
I guess this is not the ideal solution. I followed the MQTT Application Note of SIM7020 and developed my own solution completely based on it. The only issue that I'm currently facing is that whenever I turn on the device, I always need to disconnect the previous connection with the broker before attempting a new connection, otherwise it won't let me pub/sub anything.

erazor83, 2023-02-28:
@capedra can you share your code somewhere?

Thanks a lot!

ameerhazo, 2023-06-06:
Hi @1mek, I'm also interested in your work. Would you be able to share your code? Thank you.

ameerhazo, 2023-06-06:
Hello @1mek, I'm also interested in the work that you've done. Could you please share the mqtt code example that you tested your code with? I would presume that you'd set the gprs flag to false because the 7020 doesn't support it right?

Kind regards,
Ameer

1mek, 2023-06-09:
Its tested with the included mqtt examples in tinygsm. mqtt over tcpsockets.

Not the sim7020mqtt commands @capedra mentioned.