vshymanskyy/TinyGSM #695: LilyGo SIM7000G + LTE + WIFI

uxigene, 2022-11-21:
[*] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[ ] Bug or issue with library functionality (ie, sending data over TCP/IP)
[*] Question or request for help

### What are you working with?

Modem: SIM7000
Main processor board: ESP32
TinyGSM version: 0.11.15

### Scenario, steps to reproduce
I have [LilyGo SIM7000G ](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G) and trying to use TinyGSM with WiFi and LTE depending on my config. I tried to define the following:

#define TINY_GSM_USE_GPRS false
#define TINY_GSM_USE_WIFI true

But nothing hapens. It gets rewriten in "Just in case someone defined the wrong thing..." section. I found out that TINY_GSM_MODEM_HAS_WIFI is missing. But when defined got 'TinyGsm {aka class TinyGsmSim7000}' has no member named 'networkConnect' error. 

So, my question is as follow. Can I make TinyGsm to work with ESP32 and SIM7000G in both LTE and WIFI configurations?

### Expected result
Sketch compiles and both LTE and WIFI works as expected.

### Actual result
Compilation error:
'TinyGsm {aka class TinyGsmSim7000}' has no member named 'networkConnect'

uxigene, 2022-11-22:
UPD: I also tried to define:
#define TINY_GSM_MODEM_ESP8266

and 

#define TINY_GSM_USE_GPRS false
#define TINY_GSM_USE_WIFI true

But in that case it won't connect to my WiFi network.

schulzjona, 2023-11-09:
Just wondering why dont you use LTE via TinsGSM and WiFi via ESP Wifi module? That work for me with the sim7000g