vshymanskyy/TinyGSM #700: Send a SD file

sebincic, 2022-12-22:
hola, estoy usando SIM7000g para enviar datos por MQTT con tinyGSM. Esto funciona bien.
Ahora, necesito enviar un archivo de texto (en una tarjeta SD) a un servidor por GPRS. alguna idea para hacer esto?

thanks

trogper, 2023-02-05:
Sorry, this is not the right place to ask such questions. This place is for issues/bugs/features of the TinyGSM library itself.
You have to implement such functionality by yourself.

droidblastnz, 2023-04-26:
Example send a text file from a SD card. This uses FTP to upload the file.

```
#include <SoftwareSerial.h>
#include <SD.h>

// Set up software serial connection to SIM7600x module
SoftwareSerial sim7600xSerial(10, 11);  // RX, TX

// Set up chip select pin for SD card
const int chipSelect = 4;

void setup() {
  // Open serial communication with computer
  Serial.begin(9600);
  while (!Serial);

  // Open software serial communication with SIM7600x module
  sim7600xSerial.begin(9600);

  // Initialize SD card
  if (!SD.begin(chipSelect)) {
    Serial.println("SD card initialization failed!");
    return;
  }

  Serial.println("Ready to send file...");
}

void loop() {
  // Open file on SD card
  File file = SD.open("example.txt");
  if (!file) {
    Serial.println("Failed to open file!");
    return;
  }

  // Get file size
  size_t fileSize = file.size();

  // Send AT commands to SIM7600x module to set up FTP connection
  sim7600xSerial.println("AT+FTPSTART");
  if (!waitForResponse("OK")) {
    Serial.println("FTP start failed!");
    file.close();
    return;
  }

  sim7600xSerial.println("AT+FTPCID=1");
  if (!waitForResponse("OK")) {
    Serial.println("FTP set CID failed!");
    file.close();
    return;
  }

  sim7600xSerial.println("AT+FTPSERV=\"ftp.example.com\"");
  if (!waitForResponse("OK")) {
    Serial.println("FTP set server failed!");
    file.close();
    return;
  }

  sim7600xSerial.println("AT+FTPUN=\"username\"");
  if (!waitForResponse("OK")) {
    Serial.println("FTP set username failed!");
    file.close();
    return;
  }

  sim7600xSerial.println("AT+FTPPW=\"password\"");
  if (!waitForResponse("OK")) {
    Serial.println("FTP set password failed!");
    file.close();
    return;
  }

  // Send AT command to upload file
  sim7600xSerial.print("AT+FTPPUTFILE=\"example.txt\",");
  sim7600xSerial.print(fileSize);
  sim7600xSerial.println(",300");

  // Wait for ">"
  if (!waitForResponse(">")) {
    Serial.println("FTP put file failed!");
    file.close();
    return;
  }

  // Send file contents to SIM7600x module
  while (file.available()) {
    sim7600xSerial.write(file.read());
  }

  // Wait for "OK"
  if (!waitForResponse("OK")) {
    Serial.println("FTP put file failed!");
    file.close();
    return;
  }

  Serial.println("File sent successfully!");

  // Close file and wait before sending another file
  file.close();
  delay(5000);
}

// Helper function to wait for a certain response from SIM7600x module
bool waitForResponse(String response) {
  unsigned long timeout = millis() + 10000;
  String receivedData = "";

  while (millis() < timeout) {
    if (sim7600xSerial.available()) {
      char receivedChar = sim7600xSerial.read();
      receivedData += receivedChar;
      if (receivedData.endsWith(response)) {
        return true;
      }
    }
  }

  return false;
}

```

or read the file contents and send in SMS

```
#include <SoftwareSerial.h>
#include <SD.h>

// Set up software serial connection to SIM7600x module
SoftwareSerial sim7600xSerial(10, 11);  // RX, TX

// Set up chip select pin for SD card
const int chipSelect = 4;

void setup() {
  // Open serial communication with computer
  Serial.begin(9600);
  while (!Serial);

  // Open software serial communication with SIM7600x module
  sim7600xSerial.begin(9600);

  // Initialize SD card
  if (!SD.begin(chipSelect)) {
    Serial.println("SD card initialization failed!");
    return;
  }

  Serial.println("Ready to send SMS...");
}

void loop() {
  // Open file on SD card
  File file = SD.open("example.txt");
  if (!file) {
    Serial.println("Failed to open file!");
    return;
  }

  // Read file contents into a String variable
  String messageText = "";
  while (file.available()) {
    char c = file.read();
    messageText += c;
  }

  // Send AT command to set SMS text mode
  sim7600xSerial.println("AT+CMGF=1");
  if (!waitForResponse("OK")) {
    Serial.println("SMS text mode failed!");
    file.close();
    return;
  }

  // Send AT command to set SMS recipient number
  sim7600xSerial.println("AT+CMGS=\"+1234567890\"");  // Replace with recipient's number
  if (!waitForResponse(">")) {
    Serial.println("SMS recipient failed!");
    file.close();
    return;
  }

  // Send message text
  sim7600xSerial.print(messageText);
  sim7600xSerial.write(26);  // Send Ctrl+Z to indicate end of message

  // Wait for "OK"
  if (!waitForResponse("OK")) {
    Serial.println("SMS send failed!");
    file.close();
    return;
  }

  Serial.println("SMS sent successfully!");

  // Close file and wait before sending another SMS
  file.close();
  delay(5000);
}

// Helper function to wait for a certain response from SIM7600x module
bool waitForResponse(String response) {
  unsigned long timeout = millis() + 10000;
  String receivedData = "";

  while (millis() < timeout) {
    if (sim7600xSerial.available()) {
      char receivedChar = sim7600xSerial.read();
      receivedData += receivedChar;
      if (receivedData.endsWith(response)) {
        return true;
      }
    }
  }

  return false;
}

```

fgnievinski, 2023-06-26:
there could be a generic FTP upload definition whose implementation would be overloaded for each different modem model.