vshymanskyy/TinyGSM #713: how to add ssl/tls to TinyGsmClient object.

mayankbytebeam, 2023-02-28:
I want to connect to secure MQTT broker and that requires me to set the certificates, Earlier I was using Wi-Fi to connect my esp32 to broker with WiFiClientSecure library but now I am planning to achieve the same thing with GSM.

I found this library easy to use and I have setted up the modem quicly, just checking the SSL/TLS part now.

For reference, I am using TTGO SIM7600 module and arduino-esp32 framework.

any help will be much appreciated.

SRGDamia1, 2023-02-28:
https://github.com/vshymanskyy/TinyGSM/blob/master/examples/HttpsClient/HttpsClient.ino

mayankbytebeam, 2023-02-28:
I have gone through the example code but couldn't figure out where to set the https certificates.

cezarg1410, 2023-03-10:
Od You still się esp32 You can achieve that with that library https://github.com/govorox/SSLClient
Combined with tinygsm and You mqtt client

mayankbytebeam, 2023-03-10:
Yeah, that works fine. do you know any alternative method/library by any chance?