vshymanskyy/TinyGSM #717: Very unstable SSL with SIM7000E

cezarg1410, 2023-03-12:
[x ] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[x ] Bug or issue with library functionality (ie, sending data over TCP/IP)
[ ] Question or request for help

### What are you working with?

Modem: SimCom 7000E
Main processor board: ESP32
TinyGSM version:Latest
```
#include "SSLClient.h"
#include "certs.h"
#include <HardwareSerial.h>

#define GSM_RX 14
#define GSM_TX 27

#define TEENSY_RX 16
#define TEENSY_TX 17

#define MODEM_PWRKEY 25

// Select your modem:
#define TINY_GSM_MODEM_SIM800
//#define CONFIG_MBEDTLS_SSL_VARIABLE_BUFFER_LENGTH 1024
#define TINY_GSM_RX_BUFFER 1024

#include <TinyGsmClient.h>
#include <ArduinoHttpClient.h>

HardwareSerial SerialAT(1);      //gsm

#define SerialMon Serial
#define TINY_GSM_DEBUG SerialMon
#define TINY_GSM_USE_GPRS true

TinyGsm modem(SerialAT);
TinyGsmClient gsm_transpor_layer(modem, 0);
SSLClient secure_presentation_layer(&gsm_transpor_layer);
HttpClient client = HttpClient(secure_presentation_layer, server, 443);

unsigned long prevFinished = 0;

void setup() {
#ifdef TINY_GSM_DEBUG
  SerialMon.begin(115200);
  delay(10);
#endif

  logTextLine("Wait...");

  SerialAT.begin(115200, SERIAL_8N1, GSM_RX, GSM_TX);
  while (!SerialAT)
    ;

  setupModem();
  secure_presentation_layer.setCACert(ca);
  secure_presentation_layer.setCertificate(client_crt);
  secure_presentation_layer.setPrivateKey(client_key);

 
}

bool sending = false;
[... 558 lines cut ...]
+CIPRXGET: 4,0,0

OK
AT+CIPSTATUS=0

+CIPSTATUS: 0,0,"TCP","SOME.IP.ADDRESS","443","CLOSED"

OK

(many times CLOSED)

AT+CIPSTATUS=0

+CIPSTATUS: 0,0,"TCP","SOME.IP.ADDRESS","443","CLOSED"

OK
[228327][E][SSLClient.cpp:110] connect(): start_ssl_client: -1
[228327][V][ssl_client.cpp:330] stop_ssl_socket(): Cleaning SSL connection.
AT+CIPCLOSE=0

ERROR
Status code: -2
Restarting modem

```

bastienboudet, 2023-05-24:
For SSL support on sim7000E you should use TINY_GSM_MODEM_SIM7000SSL instead of TINY_GSM_MODEM_SIM7000. There is an implementation in the exemple section

cezarg1410, 2023-05-24:
I've tried  TINY_GSM_MODEM_SIM7000SSL too without success - actually very similar errors occured.
Anyway i throw into the trash SIM7000E and replaced it with SIM7600G.
And now it works as it should - without any issues

closed 2023-12-14 by cezarg1410