vshymanskyy/TinyGSM #721: How to handle incoming call?

untitledlt, 2023-03-29:
[X] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[X] Question or request for help

### What are you working with?

Modem: SIM800L
Main processor board: ESP32
TinyGSM version: 0.11.5

In docs/examples i see how to init outgoing call but nothing about incoming calls.
When I receive incoming call, i can see a log
```
+CLIP: "+[my_phone_number]",145,"",0,"",0
[1481566] ### Unhandled: RING
```
But i can not find any information how to handle this.

P.S. I know that incoming call will break active GPRS session and I'm fine with that

droidblastnz, 2023-04-26:
Example I use to pick up a incoming call and do something....

```
if (SerialAT.available()) {
  String response = SerialAT.readString();
  if (response.indexOf("RING") != -1) {
    // Incoming call detected
    String phoneNumber = response.substring(response.indexOf("MISSED_CALL: ") + 14);  // Retrieve the phone number of the incoming call
    SerialMon.println(phoneNumber);
    phoneNumber.trim();  // Trim leading/trailing spaces
    // Extract the phone number from the response
    int colonIndex = phoneNumber.indexOf(":");
    if (colonIndex != -1) {
      phoneNumber = phoneNumber.substring(colonIndex + 1);
    }
    SerialMon.print("Incoming call from: ");
    SerialMon.println(phoneNumber);

    //Call the appropriate function based on the received DTMF tones if you have decoder chip onboard
    //Do something();

    // Hang up the call
    SerialAT.println("AT+CHUP");  // Hang up the call
  }
}
```

untitledlt, 2023-04-27:
@droidblastnz thanks for an answer but I don't think I can read serial directly as it will capture all messages from modem and it will break TinyGSM functionality

droidblastnz, 2023-04-28:
> @droidblastnz thanks for an answer but I don't think I can read serial directly as it will capture all messages from modem and it will break TinyGSM functionality

Correct maybe detail what you are trying to do?

Logs below is the incoming call where it drops the network

```

2:19:06.194 -> Incoming call from: 0---------
12:19:06.705 -> AT+CGREG?
12:19:06.705 -> 
12:19:06.746 -> OK
12:19:06.947 -> 
12:19:06.947 -> MISSED_CALL: 00:19AM 0----------
12:19:07.017 -> 
12:19:07.017 -> +CIPRXGET: 1,0
12:19:07.712 -> Network disconnected
12:19:07.712 -> AT+CSQ
12:19:07.712 -> 
12:19:07.712 -> +CSQ: 18,99
12:19:07.747 -> 
12:19:07.747 -> OK
12:19:07.747 -> AT+CGREG?
12:19:07.783 -> 
12:19:07.783 -> +CGREG: 0,1
12:19:07.783 -> 
12:19:07.815 -> OK
12:19:07.815 -> AT+CGREG?
12:19:07.815 -> 
12:19:07.815 -> +CGREG: 0,1
12:19:07.851 -> 
12:19:07.851 -> OK
12:19:07.851 -> Network re-connected
12:19:07.851 -> AT+NETOPEN?
12:19:07.883 -> 
12:19:07.883 -> +NETOPEN: 1
```