vshymanskyy/TinyGSM #723: MQTT disconnects around 24hrs and doesnt reconnect

droidblastnz, 2023-04-07:
[x] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[x] Bug or issue with library functionality (ie, sending data over TCP/IP)
[x] Question or request for help

### What are you working with?
LILYGO® TTGO T-SIM7600 ESP32 LTE Cat4/1 4G Development Board SIM7600G-H R2 
Modem: SIM7600
Main processor board: ESP32-WROVER
TinyGSM version: 0.11.5
Code: MQTT example
Broker is hivemq.com
PubSubClient version 2.8.0

### Scenario, steps to reproduce
Stay Connected to MQTT broker.hivemq.com

### Expected result
To still be connected to broker.hivemq.com

### Actual result
Run the base code (MQTTClient example) and SIM for close to 24hrs then === MQTT NOT CONNECTED ===

### Debug and AT command log
Enabled debugging

My code is working as expected, except a MQTT disconnect around 24hrs.

Using the example MQTT code I can connect to broker.hivemq.com and blink the LED without issues but after close to 24hrs disconnects and will not reconnect.

Netlight is flashing at 200ms ON/OFF registered on 4G network (+CIPCLOSE: 1,0,0,0,0,0,0,0,0,0)

Up to the disconnect the following is true:

Netlight is still flashing at 200ms ON/OFF registered on 4G network (+CIPCLOSE: 0,0,0,0,0,0,0,0,0,0)

Serial Modem Logs

10:29:56.921 -> AT+CIPRXGET=4,0
10:29:56.921 -> 
10:29:56.921 -> +CIPRXGET: 4,0,0
10:29:56.974 -> 
10:29:56.974 -> OK
10:29:56.974 -> AT+CIPCLOSE?
10:29:56.974 -> 
10:29:56.974 -> +CIPCLOSE: 0,0,0,0,0,0,0,0,0,0
10:29:57.021 -> 
10:29:57.021 -> OK
10:29:57.021 -> === MQTT NOT CONNECTED ===
10:29:57.122 -> AT+CGREG?
10:29:57.122 -> 
10:29:57.122 -> +CGREG: 0,1
10:29:57.157 -> 
10:29:57.157 -> OK
10:29:57.157 -> === MQTT NOT CONNECTED ===
10:29:57.275 -> AT+CGREG?
10:29:57.275 -> 
10:29:57.275 -> +CGREG: 0,1
10:29:57.321 -> 
10:29:57.321 -> OK
10:29:57.321 -> === MQTT NOT CONNECTED ===
10:29:57.423 -> AT+CGREG?
10:29:57.423 -> 
10:29:57.423 -> +CGREG: 0,1
10:29:57.475 -> 
10:29:57.475 -> OK

Using the Tiny_GSM library MQTT doesnt reconnect.

Tried to add a delay from 100-10000 and/or TINY_GSM_YIELD_MS 2 no resolve.

if (!mqtt.connected()) {
SerialMon.println("MQTT NOT CONNECTED!");
// Reconnect every 10 seconds
uint32_t t = millis();
if (t - lastReconnectAttempt > 10000L) {
lastReconnectAttempt = t;
if (mqttConnect()) {
lastReconnectAttempt = 0;
}
}
//delay(10000); //delay for 10 seconds
//delay(1000);
delay(100); //delay for 100 milliseconds 0.1 of a second
//delay(TINY_GSM_YIELD_MS); // Add this line to yield to the TinyGSM library 2ms
return;
}
mqtt.loop(); //Calls mqtt.loop in the pubsubclient library

T-SIM7600G-H works as expected via the uploaded code until MQTT disconnects around 24hrs. Hard to confirm the exact time and logs.

Baud rate for Modem and ESP set to 115200, cannot seem to set to any other speed and have comms to modem.

alex9087, 2023-04-09:
I HAVE THE SAME PROBLEM

droidblastnz, 2023-04-09:
> I HAVE THE SAME PROBLEM

Reading this link [https://github.com/knolleary/pubsubclient/issues/239](url)

The actual value of the Keep Alive is application specific; typically this is a few minutes. 
The maximum value is 18 hours 12 minutes and 15 seconds.

What are you using device wise and can you confirm its still network connected?

droidblastnz, 2023-04-10:
Still reading but another hit of mqtt.disconnect(); but after a pub in my case trying disconnect during the reconnect function.
https://github.com/knolleary/pubsubclient/issues/435#issuecomment-450768516

`while (processed_size < total_recv_size) {
    ...
    // try to connect to the topic
    while (!mqtt.connect(mqtthelper.topic_str)) {
        SerialMon.println(" === MQTT NOT CONNECTED === ");
        delay(100);
    }
    mqtt.publish(mqtthelper.topic_str, payload, payload_size + TIMESTAMP_BYTES);
    mqtt.disconnect();     // disconnect explicitly
    ...
}`

droidblastnz, 2023-04-10:
So the final configuration for the Test 1:
TinyGsmClient.h                                       //version 0.11.3 
PubSubClient.h                                        //version 2.8.0

PubSubClient.h as per https://pubsubclient.knolleary.net/api#setKeepAlive 15s

// MQTT_KEEPALIVE : keepAlive interval in Seconds. Override with setKeepAlive()
#ifndef MQTT_KEEPALIVE
#define MQTT_KEEPALIVE 15
#endif

PubSubClient.cpp _client->stop(); still enabled

boolean PubSubClient::loop() {
    if (connected()) {
        unsigned long t = millis();
        if ((t - lastInActivity > this->keepAlive*1000UL) || (t - lastOutActivity > this->keepAlive*1000UL)) {
            if (pingOutstanding) {
                this->_state = MQTT_CONNECTION_TIMEOUT;
                _client->stop();
                return false;

Tiny_GSM added mqtt.disconnect to mqtt,connected function

    if (!mqtt.connected()) {
        SerialMon.println("MQTT NOT CONNECTED! ");
        **mqtt.disconnect(); // Disconnect from MQTT  //added 08/04/2023**
        SerialMon.print("Network State is: ");
        SerialMon.println(modem.isGprsConnected()); //added 08/04/2023
        // Reconnect every 10 seconds
        uint32_t t = millis();
        if (t - lastReconnectAttempt > 10000L) {
            lastReconnectAttempt = t;
            if (mqttConnect()) {
                lastReconnectAttempt = 0;
            }
        }
        delay(100);
        return;
    }

Of note added time sync between modem and network

  //Force Modem to sync time with network //added 09/04/2023 TinyGSM/issues/574
  modem.sendAT("+CTZU=1"); //CTZU=1 forcing time syncronization with network, automatic time and time zone update via NITZ
  modem.waitResponse(1000);
  Serial.println("CTZU Sync'ed time with network");

droidblastnz, 2023-04-12:
To me its not MQTT disconnecting its the network.

AT+CGREG?

+CGREG: 0,1

OK
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 0,0,0,0,0,0,0,0,0,0

OK
MQTT NOT CONNECTED! 
AT+CIPSEND=0,2

+CIPERROR: 2

ERROR
AT+CIPCLOSE=0

+CIPCLOSE: 0,4

ERROR
Network State is: AT+NETOPEN?

+NETOPEN: 0

droidblastnz, 2023-04-25:
> What was youre result? I think than this is no the answer, i just added de mqtt.disconnect(), before to connect

did this resolve your issue?

closed 2023-05-01 by droidblastnz