vshymanskyy/TinyGSM #725: It takes 15 minutes to waitForNetwork()

AB-informatica-service, 2023-04-23:
Hi, I have a strange problem:

I live in Italy and use TruePhone for this IoT project.
TruePhone must be used when roaming.

The code I use is very simple:
`
        Serial.print("/nModem Info: ");
        Serial.println(gsmModem->getModemInfo());
        delay(100);

        Serial.print("SIM ICCID request: ");
        Serial.println(gsmModem->getSimCCID());
        delay(1000);

        Serial.print("Stato di registrazione: ");
        Serial.println(gsmModem->getRegistrationStatus());
        delay(100);

       Serial.print("Waiting for network...");
        while (!gsmModem->waitForNetwork())
        {
            Serial.println(" fail");
            // delay(5000);
            //  break;
        }
        Serial.println(" success");

        Serial.print("Stato di registrazione: ");
        Serial.println(gsmModem->getRegistrationStatus());
        delay(100);

       Serial.print(F("Connecting to "));
        Serial.println(GSM_APN);

        if (!gsmModem->gprsConnect(GSM_APN, GSM_USER, GSM_PASS))
        {
            Serial.println("GPRS Connect: fail");
            delay(10000);
            break;
        }
        if (gsmModem->isNetworkConnected())
        {
            Serial.println("Network connected");
        }
        Serial.println(" success");

        if (gsmModem->isGprsConnected())
        {
            Serial.println("GPRS connected");
        }
        Serial.println(gsmModem->getLocalIP());

        Serial.print("Qualità segnale: ");
        Serial.println(gsmModem->getSignalQuality());
        // Serial.print("GSM data Time: ");
        // Serial.println(gsmModem->NTPServerSync(ntpServer, 20));
        getGPSPosition();
        break;

`

And my serial monitor is:
`
Setting up AbiotClient
Avvio connessione GSM....
Il modem è spento, lo accendo.
Ho tenuto su il pin ON per 1200ms
/nModem Info: ATI  SIM7000E R1351
SIM ICCID request: +CPIN: READY
Waiting for network... fail
 fail
 fail
 fail
 fail
 fail
 fail
 fail
 fail
 fail
 fail
 fail
 fail
 fail
 fail
 fail
 success
Stato di registrazione: 5
Connecting to iot.truphone.com
Network connected
 success
GPRS connected
10.136.73.51
Qualità segnale: 20
Requesting current GPS/GNSS/GLONASS location
Couldn't get GPS/GNSS/GLONASS location, retrying in 15s.
`

As you can see, he makes many attempts, in this case 17 but in other cases 13 or 25... He's not stable.
But I would like to understand how to make it work the first time.

Thank you

AB-informatica-service, 2023-04-23:
Update:
`Serial.print("Waiting for network...");
        while (!gsmModem->waitForNetwork())
        {
            Serial.println(" fail");
            delay(2000);

            Serial.print("Stato di registrazione: ");
            Serial.println(gsmModem->getRegistrationStatus());
            Serial.println(gsmModem->getPreferredModes());
            Serial.println(gsmModem->getNetworkModes());
            Serial.println("NetworkMode:");
            Serial.println(gsmModem->getNetworkMode());
            //  break;
        }
        Serial.println(" success");`

/nModem Info: ATI  SIM7000E R1351
SIM ICCID request: *********************
Waiting for network... fail
Stato di registrazione: 2
 ((1-Cat-M),(2-NB-IoT),(3-Cat-M And NB-IoT))
 ((2-Automatic),(13-GSM Only),(38-LTE Only),(51-GSM And LTE Only))
NetworkMode:
2
 fail
Stato di registrazione: 2
 ((1-Cat-M),(2-NB-IoT),(3-Cat-M And NB-IoT))
 ((2-Automatic),(13-GSM Only),(38-LTE Only),(51-GSM And LTE Only))
NetworkMode:
2
 fail
Stato di registrazione: 2
 ((1-Cat-M),(2-NB-IoT),(3-Cat-M And NB-IoT))
 ((2-Automatic),(13-GSM Only),(38-LTE Only),(51-GSM And LTE Only))
NetworkMode:
2
 fail
Stato di registrazione: 2
 ((1-Cat-M),(2-NB-IoT),(3-Cat-M And NB-IoT))
 ((2-Automatic),(13-GSM Only),(38-LTE Only),(51-GSM And LTE Only))
NetworkMode:
2
 fail
Stato di registrazione: 2
 ((1-Cat-M),(2-NB-IoT),(3-Cat-M And NB-IoT))
 ((2-Automatic),(13-GSM Only),(38-LTE Only),(51-GSM And LTE Only))
NetworkMode:
2
 fail
Stato di registrazione: 2
 ((1-Cat-M),(2-NB-IoT),(3-Cat-M And NB-IoT))
 ((2-Automatic),(13-GSM Only),(38-LTE Only),(51-GSM And LTE Only))
NetworkMode:
2
 fail
Stato di registrazione: 2
 ((1-Cat-M),(2-NB-IoT),(3-Cat-M And NB-IoT))
 ((2-Automatic),(13-GSM Only),(38-LTE Only),(51-GSM And LTE Only))
NetworkMode:
2
 fail
Stato di registrazione: 2
 ((1-Cat-M),(2-NB-IoT),(3-Cat-M And NB-IoT))
 ((2-Automatic),(13-GSM Only),(38-LTE Only),(51-GSM And LTE Only))
NetworkMode:
2
 fail
Stato di registrazione: 2
 ((1-Cat-M),(2-NB-IoT),(3-Cat-M And NB-IoT))
 ((2-Automatic),(13-GSM Only),(38-LTE Only),(51-GSM And LTE Only))
NetworkMode:
2
 fail
Stato di registrazione: 2
 ((1-Cat-M),(2-NB-IoT),(3-Cat-M And NB-IoT))
 ((2-Automatic),(13-GSM Only),(38-LTE Only),(51-GSM And LTE Only))
NetworkMode:
2
 fail
Stato di registrazione: 2
 ((1-Cat-M),(2-NB-IoT),(3-Cat-M And NB-IoT))
 ((2-Automatic),(13-GSM Only),(38-LTE Only),(51-GSM And LTE Only))
NetworkMode:
2
 fail
Stato di registrazione: 2
 ((1-Cat-M),(2-NB-IoT),(3-Cat-M And NB-IoT))
 ((2-Automatic),(13-GSM Only),(38-LTE Only),(51-GSM And LTE Only))
NetworkMode:
2
 fail
Stato di registrazione: 2
 ((1-Cat-M),(2-NB-IoT),(3-Cat-M And NB-IoT))
 ((2-Automatic),(13-GSM Only),(38-LTE Only),(51-GSM And LTE Only))
NetworkMode:
2
 fail
Stato di registrazione: 3
 ((1-Cat-M),(2-NB-IoT),(3-Cat-M And NB-IoT))
 ((2-Automatic),(13-GSM Only),(38-LTE Only),(51-GSM And LTE Only))
NetworkMode:
2
 fail
Stato di registrazione: 3
 ((1-Cat-M),(2-NB-IoT),(3-Cat-M And NB-IoT))
 ((2-Automatic),(13-GSM Only),(38-LTE Only),(51-GSM And LTE Only))
NetworkMode:
2
 fail
Stato di registrazione: 3
 ((1-Cat-M),(2-NB-IoT),(3-Cat-M And NB-IoT))
 ((2-Automatic),(13-GSM Only),(38-LTE Only),(51-GSM And LTE Only))
NetworkMode:
2
 fail
Stato di registrazione: 0
 ((1-Cat-M),(2-NB-IoT),(3-Cat-M And NB-IoT))
 ((2-Automatic),(13-GSM Only),(38-LTE Only),(51-GSM And LTE Only))
NetworkMode:
2
 fail
Stato di registrazione: 0
 ((1-Cat-M),(2-NB-IoT),(3-Cat-M And NB-IoT))
 ((2-Automatic),(13-GSM Only),(38-LTE Only),(51-GSM And LTE Only))
NetworkMode:
2
 fail
Stato di registrazione: 0
 ((1-Cat-M),(2-NB-IoT),(3-Cat-M And NB-IoT))
 ((2-Automatic),(13-GSM Only),(38-LTE Only),(51-GSM And LTE Only))
NetworkMode:
2
 fail
Stato di registrazione: 0
 ((1-Cat-M),(2-NB-IoT),(3-Cat-M And NB-IoT))
 ((2-Automatic),(13-GSM Only),(38-LTE Only),(51-GSM And LTE Only))
NetworkMode:
2
 fail
Stato di registrazione: 0
 ((1-Cat-M),(2-NB-IoT),(3-Cat-M And NB-IoT))
 ((2-Automatic),(13-GSM Only),(38-LTE Only),(51-GSM And LTE Only))
NetworkMode:
2
 fail
Stato di registrazione: 0
 ((1-Cat-M),(2-NB-IoT),(3-Cat-M And NB-IoT))
 ((2-Automatic),(13-GSM Only),(38-LTE Only),(51-GSM And LTE Only))
NetworkMode:
2

droidblastnz, 2023-04-25:
How good is the signal?  My board SIM7600 details this... if its was first time registration?

[https://github.com/vshymanskyy/TinyGSM](url)

Failed connection or no data received

The first connection with a new SIM card, a new module, or at a new location/tower may take a LONG time - up to 15 minutes or even more, especially if the signal quality isn't excellent. If it is your first connection, you may need to adjust your wait times and possibly go to lunch while you're waiting

```

  // 0 – - 113 dBm or less
  // 1 – - 111 dBm
  // 2...30 – - 109... - 53 dBm
  // 31 – - 51 dBm or greater
  // 99 – not known or not detectable
  // 100 – - 116 dBm or less
  // 101 – - 115 dBm
  // 102…191 – - 114... - 26dBm
  // 191 – - 25 dBm or greater
  // 199 – not known or not detectable
  // 100…199 – expand to TDSCDMA, indicate RSCP received
  //Bit Error Rate %
  // 0 – <0.01%
  // 1 – 0.01% --- 0.1%
  // 2 – 0.1% --- 0.5%
  // 3 – 0.5% --- 1.0%
  // 4 – 1.0% --- 2.0%
  // 5 – 2.0% --- 4.0%
  // 6 – 4.0% --- 8.0%
  // 7 – >=8.0%
  // 99 – not known or not detectable
```

AB-informatica-service, 2023-04-25:
Hi, I solved.

I set:
        gsmModem->setNetworkMode(38);
        delay(100);
        gsmModem->setPreferredMode(1);
        delay(100);

 And now it connects right away.

droidblastnz, 2023-04-25:
Pretty sure you will be best to set network to automatic (2) Read it
somewhere in tinygsm notes. Guessing in your case automatic (2) is causing
the delay as it works through finding available networks?

TruePhone must be used when roaming. What happens when LTE isn't available
if roaming?

AB-informatica-service, 2023-04-25:
We have 4G in every area. And above all we need speed and when we fix this device in one place, it stays there for years motionless (not moving)

closed 2023-04-25 by AB-informatica-service