vshymanskyy/TinyGSM #737: Handling +CIPEVENT in SIM7600

protttoy, 2023-07-24:
[x] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?
[x] Bug or problem compiling the library

### What are you working with?

Modem: SIMCOM SIM7600G-H
Main processor board: ESP32 
TinyGSM version: 0.11.5 

In TinyGsmClientSIM7600.h , class TinyGsmSim7600, function waitResponse() , line 813

```
else if (data.endsWith(GF("+CIPEVENT:"))) {
          // Need to close all open sockets and release the network library.
          // User will then need to reconnect.
          DBG("### Network error!");
          if (!isGprsConnected()) { gprsDisconnect(); }
          data = "";
        }
```

from the datasheet:

> +CIPEVENT: NETWORK CLOSED UNEXPECTEDLY
> Network is closed for network error(Out of service, etc). When this event happens, user‟s application needs to check and close all opened sockets, and then uses AT+NETCLOSE to release the network library if AT+NETOPEN? shows the network library is still opened.

Should the line  if (!isGprsConnected()) be  if (isGprsConnected())  ?

JU-Spec-Dev, 2023-09-20:
I Agree, your fix makes sense. It might explain some of the issues I've been having.
 The sim 5360 has the same code error.

fabiopolar, 2023-09-20:
If the door is not opened, close the door.
Have several SIM7600 TPCI-E boards here, and having network suddenly disconnect errors, I will try this FIX.

JU-Spec-Dev, 2023-09-21:
> If the door is not opened, close the door. Have several SIM7600 TPCI-E boards here, and having network suddenly disconnect errors, I will try this FIX.

How are you handling the disconnects, appart from the library.

I'm experiencing my modems shuting down completely. Both LEDs connected to the modem switchoff.
I'm assuming it happens when the network drops and the library doesn't handle it so the micro connected to it continues trying to use the modem as if connected.

What exactly are you experiencing on a sudden network disconnect?

fabiopolar, 2023-09-21:
Hi,

I took a dirty path to avoid those dead ends:

`bool isConnected = false;
  for (int i = 0; i <= 4; i++) {

    uint8_t network[] = {
        2,  //Automatic/
        13, //GSM only/
        38, //LTE only/
        51  //GSM and LTE only/
    };
    Serial.printf("Try %d method\n", network[i]);
    modem.setNetworkMode(network[i]);
    delay(3000);
    
    int tryCount = 60;
    while (tryCount--) {
      if(tryCount == 0){
        ESP.restart();
      }
      int16_t signal =  modem.getSignalQuality();
      Serial.print("Signal: ");
      Serial.print(signal);
      Serial.print(" ");
      Serial.print("isNetworkConnected: ");
      isConnected = modem.isNetworkConnected();
      Serial.println( isConnected ? "CONNECT" : "NO CONNECT");
      if (isConnected) {
        break;
      }
      delay(1000);
      digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    }
    if (isConnected) {
        break;
    }
  }`

The first one is on setNetwork mode fail.

The next one is on MQTT connect function (pubsubclient Lib + Tinygsm).

`if (!mqtt.connected()) {

      SerialMon.println("=== MQTT NOT CONNECTED ===");

      uint32_t t = millis();
      if (t - lastReconnectAttempt > 60000L) {
        modem.restart();
        ESP.restart();
      }

      if (t - lastReconnectAttempt > 10000L) {
        lastReconnectAttempt = t;
        if (mqttConnect()) {
          lastReconnectAttempt = 0;
        }
      }
      delay(100);
      return;
    }`