vshymanskyy/TinyGSM #743: How to maintain persistent HTTPS connection

AaronConvery, 2023-08-14:
[ ] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?
 Question or request for help

### What are you working with?

Modem: SIM7600CE-T
Main processor board: ESP32 - WROVER -IE
TinyGSM version: 0.11.7
Code: All sensitive data is redacted
#define TINY_GSM_MODEM_SIM7600

#define SerialMon Serial
#define SerialAT Serial1

#define DUMP_AT_COMMANDS

#define TINY_GSM_DEBUG SerialMon

#define TINY_GSM_TEST_GPRS true
#define TINY_GSM_TEST_TCP true

#define GSM_PIN ""

const char apn[] = "REDACTED";
const char gprsUser[] = "REDACTED";
const char gprsPass[] = "REDACTED";

const char server[] = "REDACTED";
const char resource[] = "REDACTED";
const int port = 443;

#include <TinyGsmClient.h>
#include <ArduinoHttpClient.h>
#include "SSLClient.h"
#include "utilities.h"
#include "certs.h"
#include <esp_int_wdt.h>
#include <esp_task_wdt.h>

#ifdef DUMP_AT_COMMANDS
#include <StreamDebugger.h>
StreamDebugger debugger(SerialAT, SerialMon);
TinyGsm modem(debugger);
#else
TinyGsm modem(SerialAT);
#endif

#define TWATCHDOG_TIMEOUT 120

TinyGsmClient base_client(modem, 0);
SSLClient secure_layer(&base_client);
HttpClient client = HttpClient(secure_layer, server, port);

void setup()
{
    SerialMon.begin(115200);
    delay(10);

    SerialAT.begin(UART_BAUD, SERIAL_8N1, MODEM_RX, MODEM_TX);

    pinMode(LED_PIN, OUTPUT);
    digitalWrite(LED_PIN, HIGH);

    pinMode(MODEM_PWRKEY, OUTPUT);
    digitalWrite(MODEM_PWRKEY, HIGH);
    delay(300);
    digitalWrite(MODEM_PWRKEY, LOW);

    pinMode(MODEM_FLIGHT, OUTPUT);
    digitalWrite(MODEM_FLIGHT, HIGH);

    secure_layer.setCACert(root_ca);
}

void light_sleep(uint32_t sec)
{
    esp_sleep_enable_timer_wakeup(sec * 1000000ULL);
    esp_light_sleep_start();
}

bool isConnected = false;

void loop()
{
    if (!isConnected) {
        DBG("Initializing modem...");
        if (!modem.init()) {
            if (!modem.restart()) {
                DBG("Failed to restart modem, delaying 10s and retrying");
                delay(10000);
                return;
            }
        }

        String ret;
        do {
            ret = modem.setNetworkMode(54);
            delay(500);
        } while (!ret);

        DBG("Waiting for network...");
        if (!modem.waitForNetwork(600000L)) {
            DBG("Network not available, sleeping for 10 seconds");
            light_sleep(10);
            return;
        }

        if (modem.isNetworkConnected()) {
            DBG("Network connected");
        }

        DBG("Connecting to", apn);
        if (!modem.gprsConnect(apn, gprsUser, gprsPass)) {
            DBG("Failed to connect to GPRS, sleeping for 10 seconds");
            light_sleep(10);
            return;
        }

        isConnected = modem.isGprsConnected();
        DBG("GPRS status:", isConnected ? "connected" : "not connected");
    }

    if (isConnected) {
        DBG("Making GET request securely...");
        client.beginRequest();
        client.get(resource);
        client.sendHeader("Connection", "keep-alive");
        client.endRequest();

        int status_code = client.responseStatusCode();
        String response = client.responseBody();

        Serial.print("Status code: ");
        Serial.println(status_code);
        Serial.print("Response: ");
        Serial.println(response);

        if (status_code < 0) {
            DBG("Error encountered. Reconnecting...");
            isConnected = false;
        }
    } else {
        DBG("Lost connection. Reinitializing...");
        delay(5000);
        return;
    }

    delay(500);
}

### Scenario, steps to reproduce
I want the HTTPS connection to remain active after first initialising in order to receive updates on the data it is trying to request

### Expected result
I want to receive live updates on the data requested in the get request

### Actual result
The connection is being established and a successful get request is made but then the connection is being closed despite me telling it to stay alive both in my header and on the server code. It is not explicitly being told to close yet it still is

### Debug and AT command log
All sensitive data has been redacted 
16:58:19.674 -> AT+CIPRXGET=2,0,63
16:58:19.707 -> 
16:58:19.707 -> +CIPRXGET: 2,0,63,2
16:58:19.707 -> [REDACTED]
16:58:19.707 -> 
16:58:19.707 -> OK
16:58:19.707 -> AT+CIPRXGET=2,0,2
16:58:19.707 -> 
16:58:19.707 -> +CIPRXGET: 2,0,2,0
16:58:19.707 -> [REDACTED]
16:58:19.707 -> 
16:58:19.707 -> OK
16:58:19.707 -> Status code: 200
16:58:19.707 -> Response: OPEN_VALVE
16:58:20.225 -> [14802] Making GET request securely...
16:58:20.225 -> AT+CIPCLOSE=0
16:58:20.225 -> 
16:58:20.225 -> +CIPCLOSE: 0,4
16:58:20.225 -> 
16:58:20.225 -> ERROR
16:58:20.225 -> AT+CIPRXGET=1
16:58:20.225 -> 
16:58:20.225 -> OK
16:58:20.225 -> AT+CIPOPEN=0,"TCP","[REDACTED]",443
16:58:20.257 -> 
16:58:20.257 -> OK
16:58:20.386 -> 
16:58:20.386 -> +CIPOPEN: 0,0
16:58:20.386 -> AT+CIPCLOSE=0
16:58:20.418 -> 
16:58:20.418 -> OK
16:59:20.416 -> Status code: -3
16:59:20.416 -> Response:

droidblastnz, 2023-08-16:
Same issue but with MQTT [](https://github.com/Xinyuan-LilyGO/T-SIM7600X/issues/40) see the last modded code hope it helps. If you are using a consumer SIM it will be disconnected mine was repeating the disconnect very 23-24hrs.

AaronConvery, 2023-08-16:
> Same issue but with MQTT [here](https://github.com/Xinyuan-LilyGO/T-SIM7600X/issues/40) see the last modded code hope it helps. If you are using a consumer SIM it will be disconnected mine was repeating the disconnect very 23-24hrs.

Mine is currently closing the connection as soon as it makes the first get request, if it was 18-24hours that would actually be somewhat manageable, did you ever encounter this or was it a fairly persistent connection straight away?

droidblastnz, 2023-08-17:
Mine was the use of a consumer SIM versus M2M SIM or at least suspected that but it now stays connected. Lots of reports re psu voltage, signal etc but I went through each of those and none where my issue. 

The only other advice is to use the native AT commands to build your case up but in some cases you cant or install a LAMP or IIS instance local and then monitor the logs to see what is happening. I needed to do this with MQTT docker and also FTP Filezilla as mu host doesnt allow me to see logs. It highlighted issues in my code now after a year of coding my IOT remote device works as it should....

SherifMEldeeb, 2024-09-05:
I get a lot of MQTT disconnection even with good signal can you give me any clue to look for 
@droidblastnz

droidblastnz, 2024-09-06:
As per my post https://github.com/vshymanskyy/TinyGSM/issues/743#issuecomment-1679886912 code is there to enforce the connection,  since I revised it hasnt been a issue.

droidblastnz, 2024-09-06:
> I get a lot of MQTT disconnection even with good signal can you give me any clue to look for @droidblastnz

As per my post https://github.com/vshymanskyy/TinyGSM/issues/743#issuecomment-1679886912 code is there to enforce the connection,  since I revised it hasnt been a issue.

droidblastnz, 2024-09-06:
> > I get a lot of MQTT disconnection even with good signal can you give me any clue to look for @droidblastnz
> 
> As per my post [#743 (comment)](https://github.com/vshymanskyy/TinyGSM/issues/743#issuecomment-1679886912) code is there to enforce the connection, since I revised it hasnt been a issue.

https://github.com/Xinyuan-LilyGO/T-SIM7600X/issues/40#issuecomment-1530046150