vshymanskyy/TinyGSM #744: TINY_GSM_RX_BUFFER is not overridden on SIM7500

yuzukihironaka-lefixea, 2023-08-21:
## What type of issues is this?

I defined `#define TINY_GSM_RX_BUFFER 1024` at the top of `main.cpp` but it is not overridden.

### What are you working with?

Modem: SIM7500JC
Main processor board: ESP32 with PlatformIO
TinyGSM version: 0.11.7

### Scenario, steps to reproduce

Define `#define TINY_GSM_RX_BUFFER 1024` on `main.cpp`.

```cpp
#define TINY_GSM_RX_BUFFER 1024

TinyGsm modem(SerialAT);
TinyGsmClient client(modem, 0);
```

Monitor `.read()` with Oscilloscope.

```cpp
while (client.available()) {
  // Monitor the execution time of .read() with IO12
  digitalWrite(12, true);
  char c = client.read();
  digitalWrite(12, false);
}
```

### Expected result

About 600~800 bytes of data is received at one time.

The waveform is as follows.

[image]

### Actual result

The data is not received one time.

The waveform is as follows, it appears that the size of RX buffer remains 64 bytes.

In My project,  The receive process needs to be complete within at least 50ms.

I've already set Baud Rate as `3686400` so I couldn't increace Transrate Rate any more.

[image]

If I edit `TinyGsmTCP.tpp`, it is comletely overridden.

```cpp
#if !defined(TINY_GSM_RX_BUFFER)
// edit here
// #define TINY_GSM_RX_BUFFER 64
#define TINY_GSM_RX_BUFFER 1024
#endif
```

But the declaration in `main.cpp` should be prioritized over the declaration in `TinyGsmTCP.tpp`.

And also I couldn't obtain a stable connection that wrapped with `PubSubClient` or `SSlClient` if I edit in this way.(I'm going to make new issue about this problem if this issue get solved)

Thank you.

animevietsub, 2023-08-21:
Make sure that you call `#define TINY_GSM_RX_BUFFER 650` before `#include<TinyGsmClient.h>`. That should be:

```
#define TINY_GSM_RX_BUFFER 650
... // Any pre-define with library TinyGsmClient
#include<TinyGsmClient.h>
...
TinyGsm modem(SerialAT);
TinyGsmClient client(modem);
...
```

yuzukihironaka-lefixea, 2023-08-21:
@animevietsub Thanks for the reply.

Yes, I call any pre-define before `#include<TinyGsmClient.h>` in order as below.

```cpp
// define TinyGsm
#define TINY_GSM_RX_BUFFER 1024
#define TINY_GSM_MODEM_SIM7500
#define SerialAT Serial1
// Pin assign
#define SIM_TX_PIN 17
#define SIM_RX_PIN 16
...
// include libraries
#include <TinyGsmClient.h>
#include <PubSubClient.h>
...
```

My understanding is that `#define TINY_GSM_RX_BUFFER 1024` is called first.

`#define` value is determined before compiling, I wonder if it has something to do with the order of compiling?