vshymanskyy/TinyGSM #755: .sendAT() and .waitResponse() function documentation/examples?

GavinBeattie, 2023-10-10:
[x] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[ ] Bug or issue with library functionality (ie, sending data over TCP/IP)
[x] Question or request for help

### What are you working with?

Modem: 

Code: 

### Scenario, steps to reproduce

### Expected result

### Actual result

### Debug and AT command log

droidblastnz, 2023-10-30:
```
#include <TinyGsmClient.h>
#include <SoftwareSerial.h>

// Define the GSM module's serial port
SoftwareSerial SerialAT(7, 8); // RX, TX (you may need to adjust the pin numbers)

// Define your APN and other GSM network parameters
const char apn[] = "your_apn";
const char gprsUser[] = "your_username";
const char gprsPass[] = "your_password";

TinyGsm modem(SerialAT);

void setup() {
  Serial.begin(9600);
  SerialAT.begin(9600); // Use the baud rate your GSM module requires
  delay(3000);

  // Restart the modem, wait for network registration
  modem.restart();
  modem.networkConnect(apn, gprsUser, gprsPass);
}

void loop() {
  // Send an AT command
  SerialAT.println("AT+CGSN"); // Query the module's IMEI

  // Wait for a response (with a timeout)
  if (modem.waitForResponse(10000)) {
    Serial.println("IMEI: " + modem.stream.readString());
  } else {
    Serial.println("No response");
  }

  delay(5000); // Wait for a moment before the next command
}
```

GavinBeattie, 2023-11-02:
modem.networkConnection --> no such member?
modem.waitForResponse --> modem.waitResponse?